CloverHackyColor/CloverBootloader #272: Stuck in Clover r5125

Airtonjunior54, 2020-10-18:
When I choose Catalina, clover stucks here. Don't know what might be.
Works perfectly in r5123. Issue started in r5124
I already added the new OC Quirks in my Config.plist

[image]

nikhil19rathore99, 2020-10-19:
Same thing happens to me please help!!

sn0wfL4ke98, 2020-10-19:
try and go 5125
5124 bugs a lot

SergeySlice, 2020-10-19:
Replace AptioMemoryFix to OpenRuntime

chatelp, 2020-10-29:
@SergeySlice I'm using OpenRuntime and I have exactly the same issue.

arabesc, 2020-10-29:
@chatelp, have you added the new quirks section to the config.plist?

chatelp, 2020-10-30:
@arabesc yes, unfortunately it changes nothing. Also, I have the latest version of OpenRuntime and Clover (r5126)

juliantoledo, 2020-11-12:
I was having issues with r5126, I was able to fix them by replacing AptioMemoryFix with OpenRuntime
and Enabling Quirks:
RebuildAppleMemoryMap, ProtectMemoryRegions

closed 2020-12-20 by SergeySlice

chatelp, 2020-12-20:
Hi @SergeySlice, I see you closed the issue, has it been corrected in a newer release? Thanks for the help.

chatelp, 2020-12-20:
I still have no idea what's happening and I've tried multiple suggested solutions, including replacing Aptio by OpenRuntime. @SergeySlice did the logs provided any insight onto what is happening? I'm still on v5120 today 👎