CloverHackyColor/CloverBootloader #273: UNKOWN CPU

bugraduz, 2020-10-18:
I am using BİG SUR beta 10. My cpu name is unkown
[image]

SergeySlice, 2020-10-19:
My CPU is known.

jesusosx, 2020-10-20:
[image]

Mismo Error

SergeySlice, 2020-10-20:
Any info?

SergeySlice, 2020-10-21:
Check if you have CPU section in config.plist. It must be present else this issue.

xma, 2020-10-21:
ok - adding an empty key CPU in config.plist solve the 'About this Mac' issue 
closed here : https://github.com/xma/T480-Clover/issues/26#issuecomment-713759606 
Thank you @SergeySlice

closed 2020-10-28 by SergeySlice