CloverHackyColor/CloverBootloader #275: Clover affects appleid

jantoonen, 2020-10-20:
After using 5125, I had issues with my appleid. Re-login for my appleid did not fix the issue. Restoring my old EFI did.....

DSPeelJ, 2020-10-20:
I have the same issue, when I check my system is after bootimg, it is blank

DSPeelJ, 2020-10-20:
A little more information for my system, Lenovo Yoga 720  17-7700HQ using MacBookPro14,1.  Booting older version of clover to Catalina 10.15.7 works fine, I've had this running since High Sierra.  On 5025, system info (from Hackintool,) shows ??? by System ID, all other SMBIOS info still populates.  Booting Opencore 0.6.3 instead of Clover works for Catalina. Framebuffer patches that have worked since High Sierra for HD630 breaks on Big Sur Beta Clover 5025 and Opencore 0.6.3 (separate issue)

SergeySlice, 2020-10-20:
Some debug info about the issue? I see no reason for AppleID to be changed.

jantoonen, 2020-10-20:
From later messages i conclude that the appleid issue is caused by a SystemID not (correctly/unchanged) filled. The appleid checks the contents of SystemID to see if you are still running on the same HW.

Regards

Jan

SergeySlice, 2020-10-21:
Is it the same macOS version? May be new system works different with SystemID?

jantoonen, 2020-10-21:
It is the same HD/installed system.

Jan

Jan

SergeySlice, 2020-10-28:
Did you set SystemID manually or hope that it will be random generated everytimes the same?

DSPeelJ, 2020-10-28:
I set the ID manually.  It is the same info I've been using since high sierra. Please let me know what other info would be helpful

DSPeelJ, 2020-10-30:
my last comment was misleading.  I have been using the following SMBIOS info since high sierra.
I haven't had the missing system id issue until this version of clover.  same with 5126. More information must be required that I'm missing but this does work with opencore.
Not sure what I'm missing.

Thanks
	<key>SMBIOS</key>
	<dict>
		<key>BiosReleaseDate</key>
		<string>06/25/2017</string>
		<key>BiosVendor</key>
		<string>Apple Inc.</string>
		<key>BiosVersion</key>
		<string>MBP141.88Z.0161.B01.1706251744</string>
		<key>Board-ID</key>
		<string>Mac-B4831Cxxxxxxxxxx</string>
		<key>BoardManufacturer</key>
		<string>Apple Inc.</string>
		<key>BoardType</key>
		<integer>10</integer>
		<key>ChassisAssetTag</key>
		<string>MacBook-Aluminum</string>
		<key>ChassisManufacturer</key>
		<string>Apple Inc.</string>
		<key>ChassisType</key>
		<string>0x09</string>
		<key>Family</key>
		<string>MacBook Pro</string>
		<key>LocationInChassis</key>
		<string>Part Component</string>
		<key>Manufacturer</key>
		<string>Apple Inc.</string>
		<key>Memory</key>
		<dict>
			<key>Modules</key>
			<array>
				<dict>
					<key>Frequency</key>
					<integer>2133</integer>
					<key>Serial</key>
					<string>12345678</string>
					<key>Size</key>
					<integer>8192</integer>
					<key>Slot</key>
					<integer>0</integer>
					<key>Type</key>
					<string>DDR4</string>
					<key>Vendor</key>
					<string>Samsung</string>
				</dict>
				<dict>
					<key>Frequency</key>
					<integer>2133</integer>
					<key>Serial</key>
					<string>87654321</string>
					<key>Size</key>
					<integer>8192</integer>
					<key>Slot</key>
					<integer>1</integer>
					<key>Type</key>
					<string>DDR4</string>
					<key>Vendor</key>
					<string>Samsung</string>
				</dict>
			</array>
		</dict>
		<key>Mobile</key>
		<true/>
		<key>ProductName</key>
		<string>MacBookPro14,1</string>
		<key>SerialNumber</key>
		<string>C0xxxxxxxxxx</string>
		<key>Version</key>
		<string>1.0</string>
	</dict>

Deeka0, 2020-11-07:
I am also facing the same issue as I upgraded clover from v5122 to v5126. I can't use iCloud services and I'm always prompted to fix problems relating my Apple ID in system preferences. What debug logs are needed?

I'm running Catalina 10.15.3
Thanks.

deverassilva, 2020-11-18:
since version 5124 to 5126 is crashing, only version 5123.1 worked after the 5122 that I had.

5124 to 5126 is crashing locking at boot

closed 2021-02-15 by SergeySlice