CloverHackyColor/CloverBootloader #276: How to use the version 5125?

wdrich, 2020-10-20:
I am using version 5122 well, OS 10.15.7. When upgrading to version 5125, a black screen starts. Is there any documentation for the new version of Clover?

error info:
[EB|#LOG:EXITBS:START]

corint1, 2020-10-20:
try to give more details ... what configuration do you have, what did you try to do and you didn't succeed, what didn't you understand ... and first of all try to read the forum for clover bootloader ....
https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/

octoforce, 2020-10-20:
You need add quirks settings into config.plist from config sample and turn on/off some items according to your computer.

octoforce, 2020-10-20:
BTW my computer is i3 9100f and rx570.
Here are my quirks settings which is changed.
AvoidRuntimeDefrag: YES

DevirtualiseMmio: YES

DisableSingleUser: NO

DisableVariableWrite: NO

DiscardHibernateMap: NO

EnableSafeModeSlide: YES

EnableWriteUnprotector: YES

ForceExitBootServices：NO

ProtectMemoryRegion：NO

ProtectSecureBoot：NO

ProtectUefiServices：NO

ProvideCustomSlide：YES

RebuildAppleMemoryMap：YES

SetupVirtualMap： NO

SignalAppleOS： NO

SyncRuntimePermissions：YES

Ales999, 2020-10-20:
Same problem, - black screen. Current not possible to load, - no more info who is bad. :-(
Trying to restore ...
ps sorry for my bad english.

corint1, 2020-10-20:
nobody cares what processors you have ... the problem is the motherboard ... the chipset ... z370? z390? plus I gave you the support forum as a reference ...

corint1, 2020-10-20:
> psps System is Mojave

for mojave no need update clover ...use the old one ...

wdrich, 2020-10-20:
I can't see the newest user guide？

Ales999, 2020-10-20:
> > psps System is Mojave
> for mojave no need update clover ...use the old one ...
Oh, last supported version for Mojave ?
I can't even imagine now how to go back to the old version :-(

MysticalOS, 2020-10-20:
> [EB|#LOG:EXITBS:START]

https://dortania.github.io/OpenCore-Install-Guide/troubleshooting/extended/kernel-issues.html#stuck-on-eb-log-exitbs-start Quirks problem. People using new clover have to also learn how to use the opencore quirks settings correctly based on their board and cpu type.

octoforce, 2020-10-20:
My motherboard is ASROCK H310cm itx

Ales999, 2020-10-20:
I'm recovery to previous clover with this post: https://www.tonymacx86.com/threads/report-issue-with-clover-4586-macos-10-13-5.255268/post-1771526

SergeySlice, 2020-10-20:
I also have to propose to change AptioMemoryFix to OpenRuntime.efi driver.

closed 2020-10-20 by SergeySlice