CloverHackyColor/CloverBootloader #281: Clover r5125 black screen after selection for the first time in 2 years with the same config

dogaoz, 2020-10-25:
Clover r5125 update dell xps 13 9350 black screen after selection for the first time in 2 years with the same config.
When I use -v verbose, I see this:

[image]

FrEaK-git, 2020-10-25:
Same for me... :(

DecadentIpsum, 2020-10-26:
Same for me...

cyberfunk, 2020-10-26:
Return to release 5122 if you want peace of mind😁

arabesc, 2020-10-26:
show your EFI folder structure and attach config.plist

denvergeeks, 2020-10-27:
> Return to release 5122 if you want peace of mind😁 Release 5123 and up introduce all kinds of issues with existing Clover setups. They're trying to integrate OpenCore I think and it's been a disaster for me. I'm sticking with 5122 until things are ironed out.

Thank you!!! Using 5122 worked for me, using this tutorial https://github.com/wmchris/DellXPS15-9550-OSX/blob/10.15/Tutorial_10.15.md and associated files at https://github.com/wmchris/DellXPS15-9550-OSX

jief666, 2020-10-28:
OpenCore integration starts at 5124, not 5123.
To jump to 5124+ you have to define a Quirks section in your config.plist with a bit more quirks than if you were using OcQuirks.efi.
Unfortunately, you can find some forum posts but no real documentation.
Ask here : https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?page=982, you'll get help.
If you're willing to help with documentation (or anything else), tell us.
Thanks.

closed 2020-10-28 by jief666

paulopera, 2020-11-03:
> OpenCore integration starts at 5124, not 5123.
> To jump to 5124+ you have to define a Quirks section in your config.plist with a bit more quirks than if you were using OcQuirks.efi.
> Unfortunately, [...]

I'm facing the same problem as the OP with r5125 and r5126 but I'm not using OpenCore.
What information do you need?
[image]

jief666, 2020-11-04:
If you have a working OpenCore or a working 5124 or 5125, follow this : https://jief-machak.gitbook.io/cloverhelp/.
If not, ask here : https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?page=982, you'll get help.