CloverHackyColor/CloverBootloader #283: GetOS Detection and Kernel Patches Booting different OS on TRX40 cpu

fabiosun, 2020-10-26:
Booting different OSX with 5125 Clover Release produces an error in GetOS detection.
In my case it detects always High Sierra also if I start my Big Sur disk (High Sierra and Big Sur are on different disk)
This on AMD platform is a big problem because kernel patches may differ on different OSX and Clover apply or not them cause bad OSX detection
In Opencore bootloader this problem does not happen in the same disk configuration:

**Big Sur partial debug log**
`64:427  0:020  **GetOSVersion: 10.13.6 (17G14033)**
64:468  0:041  Filtering KextPatches:
64:513  0:044   - [00]: com.apple.driver.AppleEthernetAquantiaAqtion (Aquantia BigSur) :: BinPatch :: [OS: 10.13.6 | MatchOS: All | MatchBuild: All] ==> allowed
64:554  0:041   - [01]: com.apple.kext.AMDSupport (Violet strips) :: BinPatch :: **[OS: 10.13.6 |** MatchOS: All | MatchBuild: All] ==> allowed
64:596  0:041  Filtering KernelPatches:
64:617  0:021   - [00]: algrey - _i386_init_slave - Remove wrmsr 0x1c8 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
64:682  0:064   - [01]: commpage_populate -remove rdmsr :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
64:724  0:041   - [02]: cpu_topology_sort -disable _x86_validate_topology :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
64:765  0:041   - [03]: cpuid_set_cache_info - cpuid 0x8000001D instead 0 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
64:807  0:041   - [04]: cpuid_set_cache_info - cpuid 0x8000001D instead 0 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
64:871  0:064   - [05]: cpuid_set_cache_info - cpuid 0x8000001D instead 4 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
64:912  0:041   - [06]: cpuid_set_cache_info - don't set cpuid_cores_per_package :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
64:953  0:041   - [07]: skip cpuid_cores_per_package test :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:018  0:064   - [08]: skip cpuid_cores_per_package test :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:059  0:041   - [09]: cpuid_set_generic_info - remove wrmsr :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:100  0:041   - [10]: cpuid_set_generic_info - set microcode=186 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:165  0:064   - [11]: cpuid_set_generic_info - set flag=1 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:206  0:041   - [12]: cpuid_set_generic_info - disable check to allow leaf7 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:248  0:041   - [13]: cpuid_set_info - GenuineIntel to AuthenticAMD  :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:312  0:064   - [14]: cpuid_set_cpufamily - force CPUFAMILY_INTEL_PENRYN :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:354  0:041   - [15]: cpuid_set_info - jmp to calculations and set cpuid_cores_per_package  :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:395  0:041   - [16]: cpuid_set_info - cores and threads calculations :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:460  0:065   - [17]: cpuid_set_info - cores and logicals count - part 1 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:502  0:041   - [18]: cpuid_set_info - cores and logicals count - part 1 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:543  0:041   - [19]: cpuid_set_info - cores and logicals count - part 2 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:585  0:041   - [20]: cpuid_set_info - cores and logicals count - part 3 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:650  0:065   - [21]: cpuid_set_info - cores and logicals count - part 3 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:691  0:041   - [22]: cpuid_set_info - cores and logicals count - part 4 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:733  0:041   - [23]: cpuid_set_info - cores and logicals count - part 4 :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:797  0:064   - [24]: i386_init - remove rdmsr :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:839  0:041   - [25]: tsc_init - remove Penryn check to execute default case :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:880  0:041   - [26]: tsc_init - remove Penryn check to execute default case :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:945  0:064   - [27]: tsc_init - grab DID and VID from MSR :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
65:986  0:041   - [28]: tsc_init - skip msr_flex_ratio test and go grab FSBFrequency from EFI :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
66:028  0:041   - [29]: lapic_init - remove version check and panic :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
66:092  0:064   - [30]: lapic_init - remove version check and panic :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
66:133  0:041   - [31]: lapic_interrupt - skip checks and prevent panic :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
66:175  0:042   - [32]: lapic_interrupt - skip checks and prevent panic :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> allowed by OS
66:240  0:064   - [33]: mtrr_update_action - fix PAT :: [OS: 10.13.6 | MatchOS: 10.13.x, 10.14.x, 10.15.x | MatchBuild: no] ==> disabled by user
66:282  0:041  Will not patch boot.efi
`

**High Sierra partial debug log**
`29:914  0:044  **GetOSVersion: 10.13.6 (17G14033)**
29:914  0:000  Filtering KextPatches:
29:914  0:000   - [00]: com.apple.driver.AppleEthernetAquantiaAqtion (Aquantia BigSur) :: BinPatch :: [OS: 10.13.6 | MatchOS: All | MatchBuild: All] ==> allowed
29:914  0:000  Filtering KernelPatches:
29:914  0:000   - [00]: algrey - _i386_init_slave - Remove wrmsr 0x1c8 :: [OS: 10.13.6 | MatchOS: 11.x | MatchBuild: no] ==> not allowed by OS
29:914  0:000   - [01]: algrey - commpage_populate -remove rdmsr :: **[OS: 10.13.6 |** MatchOS: 10.13.x,10.14.x,10.15.x,11.x | MatchBuild: no] ==> allowed by OS
29:914  0:000   - [02]: algrey - cpu_topology_sort -disable _x86_validate_topology :: [OS: 10.13.6 | MatchOS: 10.13.x,10.14.x,10.15.x,11.x | MatchBuild: no] ==> allowed by OS
29:914  0:000   - [03]: algrey - cpuid_set_cache_info - cpuid 0x8000001D instead 0 - 10.15/10.16 :: [OS: 10.13.6 | MatchOS: 10.15.x,11.x | MatchBuild: no] ==> not allowed by OS
29:914  0:000   - [04]: algrey - cpuid_set_cache_info - cpuid 0x8000001D instead 0 - 10.13/10.14 :: [OS: 10.13.6 | MatchOS: 10.13.x,10.14.x | MatchBuild: no] ==> allowed by OS
29:914  0:000   - [05]: algrey - cpuid_set_cache_info - cpuid 0x8000001D instead 4 :: [OS: 10.13.6 | MatchOS: 10.13.x,10.14.x,10.15.x,11.x | MatchBuild: no] ==> allowed by OS
29:914  0:000   - [06]: algrey - cpuid_set_cache_info - don't set cpuid_cores_per_package :: [OS: 10.13.6 | MatchOS: 10.13.x,10.14.x,10.15.x,11.x | MatchBuild: no] ==> allowed by OS
29:915  0:000   - [07]: NoOne - skip cpuid_cores_per_package test - 10.15/10.16 :: [OS: 10.13.6 | MatchOS: 10.15.x,11.x | MatchBuild: no] ==> not allowed by OS
29:915  0:000   - [08]: algrey - skip cpuid_cores_per_package test - 10.13/10.14 :: [OS: 10.13.6 | MatchOS: 10.13.x,10.14.x | MatchBuild: no] ==> allowed by OS
29:915  0:000   - [09]: algrey - cpuid_set_generic_info - remove wrmsr :: [OS: 10.13.6 | MatchOS: 10.13.x,10.14.x,10.15.x,11.x | MatchBuild: no] ==> allowed by OS
[... cut ...]
29:915  0:000   - [32]: algrey - lapic_interrupt - skip checks and prevent panic - 10.15/10.16 :: [OS: 10.13.6 | MatchOS: 10.15.x,11.x | MatchBuild: no] ==> not allowed by OS
29:915  0:000   - [33]: algrey - lapic_interrupt - skip checks and prevent panic - 10.13/10.14 :: [OS: 10.13.6 | MatchOS: 10.13.x,10.14.x | MatchBuild: no] ==> allowed by OS
29:915  0:000   - [34]: Shaneee - mtrr_update_action - fix PAT :: [OS: 10.13.6 | MatchOS: 10.13.x,10.14.x,10.15.x,11.x | MatchBuild: no] ==> allowed by OS`

thank you

fabiosun, 2020-10-27:
to complete my previous debug, attached is how OpenCore debug correctly different boots in different OS
[image]

jief666, 2020-10-28:
OpenCore use kernel version, we use macOS version.
It happened that preboot partition seems to be mixed up. If look in it, you'll find a SystemVersion.plist with 10.13.6 version in it.
We don't know why. It's an Apple bug, I think.
We'll probably wait to see if Apple fix that bug. In the mean time, you can put 11.0 for ProductVersion in /Volumes/{preboot BS}/{GUID}/System/Library/CoreServices/SystemVersion.plist
```
	<key>ProductVersion</key>
	<string>11.0</string>
	<key>ProductUserVisibleVersion</key>
	<string>11.0</string>
```

fabiosun, 2020-10-28:
thank you for your fast answer 

I see this in my preboot.. if I am not wrong and if I am not missing your explanation
Thank you again
[image]

fabiosun, 2020-10-28:
[image]
@jief666 I have studied
maybe you closed too much soon as apple bug?

jief666, 2020-10-28:
Maybe !
It's because we had that issue, same symptom, but in your case, maybe a different cause.
I didn't know there was a 11.0.1.
Send me a zip of your preboot partition, please.

fabiosun, 2020-10-28:
https://we.tl/t-AyItfAFCdG
here you go thanks
I would like also specify that it is not a problem from latest Big Sur beta..I have had this bug from first your clover I have tested (V5124) and with previous beta (10,9,8)

SergeySlice, 2021-03-16:
Is it resolved in 5131?

jief666, 2021-03-16:
@fabiosun : Looks like I completely missed your answer. I was also out 3 months out of the last 4...
Do you still have a preboot partition that cannot be boot by Clover because of OS version detection ? If yes, can you resend it.

SergeySlice, 2021-03-29:
Nobody saw this bug since october 2020.

closed 2021-03-29 by SergeySlice

fabiosun, 2021-03-29:
Sorry @SergeySlice i am not using anymore Clover and i can test anymore this bug
correct to close it :)

fabiosun, 2021-06-27:
I would like to reopen this If possible
booting with clover in MotereyBeta 1 installation :
⌘  Powered by Clover revision: 5137 (master, commit cc561f87d)

I have this : 
25:655  0:037  GetOSVersion: 11.5 (20G5042c)

so problem seems to be always there

jief666, 2021-06-28:
It's because it version 12. We have to add it in Clover.
Could you open another issue ? Like "Version 12 is not detected".

fabiosun, 2021-06-28:
I am investigated better because I do not understand for now why it is not detecting well in some condition
in other it works
I have had the same problem in the past before quitting to use clover (it was clover 5124)
If I had news I will inform you

fabiosun, 2021-06-28:
https://github.com/CloverHackyColor/CloverBootloader/issues/424