CloverHackyColor/CloverBootloader #286: MatchOS in Kext to Patch and Big Sur beta 11.0.1

fabiosun, 2020-10-29:
[image]

It seems that 11.x or 11.0.x are not working anymore.
Also if I leave empty MatchOS value that patch for Aquantia is not applied well

I can't see anymore in debug log this:

Starting Clover revision: 5125 (master, commit 6a96d4833) on American Megatrends EFI
KextsToPatch: 1 requested
 - [00]: com.apple.driver.AppleEthernetAquantiaAqtion (Aquantia BigSur) :: BinPatch :: data len: 6

with the commit shown in attach I can't find KextToPatch

closed 2020-10-29 by fabiosun

fabiosun, 2020-10-29:
I think it wasn't a bug...but a big bug solved from previous version
We could declare a kext patch in a bad way (without putting complete kext name and clover applied it only with a find/replace to all that matched that data)

thank you to have solved this (I think bug is present In latest release 5125 . using commit d0388ee7a solves this and we have to use a correct naming to have the patch applied.
In debug log I can't see any info about this Kext to patch applied and maybe also this commit is missing something

jief666, 2020-10-29:
By reading you, I can't understand if it's working or not. You said fixed, but you also said it's not applied.
We also said that version with 3 numbers are not supported yet.