CloverHackyColor/CloverBootloader #287: stuck at root

Makeroot911, 2020-10-30:
[image]

I tried clover version 5123 and above, but it's still like that
I tried it in catalina and big sur
sorry I'm not good at English

my specs :
acer e5-475g
intei i5 7200u
ram 16 gb
nvidia gt940mx
ssd adata xpg 8200pro 512gb

SergeySlice, 2020-11-04:
Try Clover 5126 and show us what Quirks you set.

ngdanghau, 2020-11-12:
I also have this problem and my Laptop is DELL E7490:  
- Intel i7 8650U  
- UHD 620 
- DW1560 (BCM94352z)

Quirks:
[image]

451725563, 2020-11-12:
I have the same problem when I use version 5126, though it is work perfectly in 5123.1.
My laptop is DELL XPS 9360
- intel i5 7200u
- UHD 620
- DW1820a
here is my Quirks:
[image]

riskyrismawan, 2020-11-19:
> I have the same problem when I use version 5126, though it is work perfectly in 5123.1.
> My laptop is DELL XPS 9360
> 
> * intel i5 7200u
> * UHD 620
> * DW1820a
>   here is my Quirks:
> 
> [image]

This happened to me too, your quirk didn't work,

OnlyWumbo, 2020-11-19:
i think i'm going back with OC again for now, before this issue fixed :(

im-v, 2020-11-23:
Did anyone found solution for this as it's happening on latest version also

Makeroot911, 2020-11-27:
after I updated to version 5127, the problem has been resolved

closed 2020-11-27 by Makeroot911