CloverHackyColor/CloverBootloader #289: GetOS empty in a clean Big Sur 11.0.1 installation from a full USB installer (TRX40 and Clover revision: 5126 (master, commit 9e7b4311c)

fabiosun, 2020-10-30:
Hi, I have create an USB full installation pen drive
Fir installation step is fine
Then when I reboot in Clover Boot menu I can choose Big Sur Preboot
I have an instant reboot because (I think) we have a problem with OSX detection and it applies all patches I have in kernel to patch and in kext to patch
`15:276  0:048  GetOSVersion: 
15:276  0:000  Filtering KextPatches:
15:276  0:000   - [00]: com.apple.driver.AppleEthernetAquantiaAqtion (Aquantia BigSur) :: BinPatch :: [OS:  | MatchOS: 11.x | MatchBuild: All] ==> allowed
15:276  0:000  Filtering KernelPatches:
15:276  0:000   - [00]: algrey - _i386_init_slave - Remove wrmsr 0x1c8 :: [OS:  | MatchOS: 10.15.x,11.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [01]: algrey - commpage_populate -remove rdmsr :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [02]: algrey - cpu_topology_sort -disable _x86_validate_topology :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [03]: algrey - cpuid_set_cache_info - cpuid 0x8000001D instead 0 - 10.15/10.16 :: [OS:  | MatchOS: 10.15.x,11.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [04]: algrey - cpuid_set_cache_info - cpuid 0x8000001D instead 0 - 10.13/10.14 :: [OS:  | MatchOS: 10.13.x,10.14.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [05]: algrey - cpuid_set_cache_info - cpuid 0x8000001D instead 4 :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [06]: algrey - cpuid_set_cache_info - don't set cpuid_cores_per_package :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [07]: NoOne - skip cpuid_cores_per_package test - 10.15/10.16 :: [OS:  | MatchOS: 10.15.x,11.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [08]: algrey - skip cpuid_cores_per_package test - 10.13/10.14 :: [OS:  | MatchOS: 10.13.x,10.14.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [09]: algrey - cpuid_set_generic_info - remove wrmsr :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [10]: algrey - cpuid_set_generic_info - set microcode=186 :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [11]: algrey - cpuid_set_generic_info - set flag=1 :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [12]: algrey - cpuid_set_generic_info - disable check to allow leaf7 :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [13]: algrey - cpuid_set_info - GenuineIntel to AuthenticAMD :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [14]: algrey - cpuid_set_cpufamily - force CPUFAMILY_INTEL_PENRYN :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [15]: algrey - cpuid_set_info - jmp to calculations and set cpuid_cores_per_package - 10.15/10.16 :: [OS:  | MatchOS: 10.15.x,11.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [16]: algrey - cpuid_set_info - cores and threads calculations - 10.15/10.16 :: [OS:  | MatchOS: 10.15.x,11.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [17]: algrey - cpuid_set_info - ryzen cores and logicals count - part 1 - 10.13 :: [OS:  | MatchOS: 10.13.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [18]: algrey - cpuid_set_info - ryzen cores and logicals count - part 1 - 10.14 :: [OS:  | MatchOS: 10.14.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [19]: algrey - cpuid_set_info - ryzen cores and logicals count - part 2 - 10.13/10.14 :: [OS:  | MatchOS: 10.13.x,10.14.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [20]: algrey - cpuid_set_info - ryzen cores and logicals count - part 3 - 10.13 :: [OS:  | MatchOS: 10.13.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [21]: algrey - cpuid_set_info - ryzen cores and logicals count - part 3 - 10.14 :: [OS:  | MatchOS: 10.14.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [22]: algrey - cpuid_set_info - ryzen cores and logicals count - part 4 - 10.13 :: [OS:  | MatchOS: 10.13.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [23]: algrey - cpuid_set_info - ryzen cores and logicals count - part 4 - 10.14 :: [OS:  | MatchOS: 10.14.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [24]: algrey - i386_init - remove rdmsr (x3) :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [25]: algrey - tsc_init - remove Penryn check to execute default case 10.14 :: [OS:  | MatchOS: 10.14.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [26]: algrey - tsc_init - remove Penryn check to execute default case :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [27]: algrey - tsc_init - grab DID and VID from MSR :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [28]: algrey - tsc_init - skip msr_flex_ratio test and go grab FSBFrequency from EFI :: [OS:  | MatchOS: All | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [29]: algrey - lapic_init - remove version check and panic - 10.15 :: [OS:  | MatchOS: 10.15.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [30]: NoOne - lapic_init - remove version check and panic - 10.16 :: [OS:  | MatchOS: 11.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [31]: algrey - lapic_init - remove version check and panic - 10.13/10.14 :: [OS:  | MatchOS: 10.13.x,10.14.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [32]: algrey - lapic_interrupt - skip checks and prevent panic - 10.15/10.16 :: [OS:  | MatchOS: 10.15.x,11.x | MatchBuild: no] ==> allowed by OS
15:276  0:000   - [33]: algrey - lapic_interrupt - skip checks and prevent panic - 10.13/10.14 :: [OS:  | MatchOS: 10.13.x,10.14.x | MatchBuild: no] ==> allowed by OS`

any advice is welcomed

by the way..a clean Catalina 10.15.7 (19H4) always from a full USB installer is working fine

SergeySlice, 2020-11-01:
Your issue is patched allowed when it shouldn't be?
I may propose to make patched NOT ALLOWED when the system is unknown.
On the other hand we have "set system version 10.10.10 when unknown". Did you look it?

fabiosun, 2020-11-01:
Thanks for your answer, I will try to detail as better I can:

**> Your issue is patched allowed when it shouldn't be?**
Yes

**> I may propose to make patched NOT ALLOWED when the system is unknown.**
Not possible for different reasons I could try to explain
AMD System to boot in OSX has the need to have some Kernel patches applied
Catalina Kernel patches could be the same or not IE Big Sur
Mojave has different and also High Sierra has different patches (Find or replace I mean)

When Clover fails to detect right OS I have chosen to boot it applies all patches or no patches...it depends from how it detects:
I do an example:
I boot in Catalina
Clover detects High Sierra
So os = High Sierra, MatchOS is HighSierra, Catalina patches are not allowed

other situation with ie Big Sur
if I create an usb pen with a full installer of Big Sur 11.0.1
first installation step is good
at the reboot after choosing BS icon to continue installation, it reboots immediately...
If you see above log OS Detection is empty and in this way we have:
OS=empty, MatchOS= all, clover applies all patches I have

if I use the great and unique clover feature (disable patches using option /  and I leave only the patches related to Big Sur , clover install well also in the second part
then, in third and forth parts we do no need any more to manually unflag not BigSur related patches because from the third installation steps Clover detect fine OS Version and it ens fine with BS installation

**> On the other hand we have "set system version 10.10.10 when unknown". Did you look it?**
No I didn't , but, if clover set to 10.10.10 if it doesn't detect well booted system..no kernel patches will be applied..and AMD system will not start

SergeySlice, 2020-11-04:
You may change 10.10.10 to any numbers you like. For example 11.0.

If a patch allowed it can't be applied if the pattern is not found.
The Hacker created this patch should make the pattern more exact matching and no need OS matching.

fabiosun, 2020-11-04:
> You may change 10.10.10 to any numbers you like. For example 11.0.
> 
> If a patch allowed it can't be applied if the pattern is not found.
> The Hacker created this patch should make the pattern more exact matching and no need OS matching.

Always thank you for your efforts and help
But, it is not possible in this way because sometimes the pattern for find is the same for different OS and it differs only for replace pattern

By the way, it is simple for me to solve using two different config..
but I think this is a bug..but I am not a developers or a Hacker
Only a simple user

Thank you again

SergeySlice, 2021-03-21:
The issue is far from modern Clover. Open new issue if any bug was noticed.

closed 2021-03-21 by SergeySlice