CloverHackyColor/CloverBootloader #290: Stuck on r5120, both r5122 and newest r5126 don't work

chatelp, 2020-10-30:
Hi ! My setup is fully functional under r5120 but I cannot migrate to the latest release before the OC merger (r5122) or the newest r5126 release.
- under r5120, this error displays at boot : `XBuffer::[] : i > _Len. System Halted` (issue [240](https://github.com/CloverHackyColor/CloverBootloader/issues/240) is closed due to "user mistake", but the mistake is never explained...)
- under r5126, I get a black screen after boot entry selection. Verbose mode displays a Kernel Panic. Comments under issue [272](https://github.com/CloverHackyColor/CloverBootloader/issues/272) indicates that I need to switch from Aptiomemoryfix to OpenRuntime, but I already have the very latest version of OpenRuntime.

My only guess is that there has been some change in the config file format since r5120, but I can't find any information in the release notes. The fact remains that this very config is fully functional under r5120 but breaks with anything newer.

Please help me :(

I'm attaching my config.plist. 
[config.plist.txt](https://github.com/CloverHackyColor/CloverBootloader/files/5465292/config.plist.txt)

SergeySlice, 2020-11-01:
Set Boot->Debug=YES
and upload /EFI/CLOVER/misc/debug.log here.

chatelp, 2020-11-01:
Here you go @SergeySlice. Thanks for taking the time to look into it. I attached the logs for r5122 and r5126.
[debug_r5126.log](https://github.com/CloverHackyColor/CloverBootloader/files/5471142/debug_r5126.log)
[debug_r5122.log](https://github.com/CloverHackyColor/CloverBootloader/files/5471143/debug_r5122.log)

SergeySlice, 2020-11-04:
You have locked CFG so you must set KernelPM=YES

chatelp, 2020-11-04:
@SergeySlice, as you can see from my attached config.plist in the first post, kernelPM is already enabled in the kernel and kext patches section. Am I missing something?

mouness2020, 2020-11-06:
save problem here 
r5126 freeze when hit enter to boot
r5125 stuck with start word

prgrmr25, 2020-11-09:
same for me it just freezes since the implementation of opencore

AndresGarciaSobrado91, 2020-11-12:
same, can't use last r5126, it gets freeze :/

juliantoledo, 2020-11-12:
I was having issues with r5126, I was able to fix them by replacing AptioMemoryFix with OcQuirks
and Enabling Quirks:
RebuildAppleMemoryMap, ProtectMemoryRegions

dadipsetdon, 2020-11-16:
Anyone resolve this? @juliantoledo, looked at OcQuirks, but I already had that instead of AptioMemoryFix, and enabling those two quirks didn't do anything for me.  My Clover is unresponsive as soon as I push a button, and I just see the bootloader screen. I did go through the USB mapping exercise, wondering if that is causing the issue since my keyboard is USB.

chatelp, 2020-11-16:
@dadipsetdon unfortunately not. It’s even the second time I’m posting a bug report about this particular problem. The first one was closed without explanation. I’d be happy if its just a configuration error from my side but, reading from all other people posting on this thread, I’m guessing there’s a real problem here. I fully understand that the devs don’t owe us anything (thanks for the great work @SergeySlice !) but I’d like to have some kind of confirmation that there is indeed a bug there so that I don’t waste any more time trying to make this work on my own :/

siliconsocket, 2020-11-17:
https://www.tonymacx86.com/threads/new-clover-release-v5-1-r5123-big-sur-support.304543/page-10

deverassilva, 2020-11-18:
since version 5124 to 5126 is crashing, only version 5123.1 worked after the 5122 that I had.

5124 to 5126 is crashing locking at boot

AndresGarciaSobrado91, 2020-11-18:
same for me @deverassilva

jief666, 2021-01-27:
Hi all,
sorry to handle this late.
I can confirm that to switch 5124 and after, you need to find the right Quirks configuration.
Anyone still having problems or you have figured it out.
If not, I'll try to help.

panosl, 2021-01-31:
> Hi all,
> sorry to handle this late.
> I can confirm that to switch 5124 and after, you need to find the right Quirks configuration.
> Anyone still having problems or you have figured it out?
> If not, I'll try to help.

It would be much easier to have a wiki page here on the project's github with the steps required in order to move from 5122 upwards. I tried to update clover twice but I've no idea what I need to do with quirks. I saw the links on tonymac, but a simple guide would be much appreciated

jief666, 2021-01-31:
If anyone is up to set a wiki page, I'd be happy to encourage.
Quirks are explained on Dortania guide. They come from OpenCore.
Take you CPU generation, like Kaby Lake, and try first to setup the quirks the same value as in the Dortania guide.
If you have a pre-5124 config.plist working, the only thing required to add to your current config.plist should be a Quirks section, like in the config.plist.
Don't change anything in your ACPI, kexts or efi drivers if you have a current working configuration.

panosl, 2021-01-31:
Thank you @jief666 I'll give it a try and I'll try to open a ticket for a README update and/or a wiki page for the basic stuff as the ones you mentioned above.

jief666, 2021-01-31:
Opening a ticket won't make that wiki page... unless you do it...
The lack of readme update is NOT a bug. It's just that nobody did it.

closed 2021-02-06 by SergeySlice