CloverHackyColor/CloverBootloader #292: get panic after updating to 5126

vincentchen666, 2020-10-31:
My laptop is running 10.15.7 with Clover 5125, it works perfect. After I updating it to 5126, it got  panic as the screenshooting.
Please Check!

[https://ftp.bmp.ovh/imgs/2020/10/774739e76b2623c8.jpg](url)

SergeySlice, 2020-11-01:
Reduce picture size and attach here. I see no it.

nooretk, 2020-11-03:
I’m also have 10.15.7 and after updating to 5126 i got Memory Panic

[image]

SergeySlice, 2020-11-04:
Use PanicNoKextDump=YES for more informative screen.
Also show your hardware specification.

nooretk, 2020-11-06:
[image]

Got this on BigSur Installer Any Fix ?
r5126

mouness2020, 2020-11-06:
i solved Memory Panic by replacing openruntime.sc
[image]
 with OsxAptioFix3Drv.efi

SergeySlice, 2020-11-09:
Different people reported different problems in one issue.
Please open own issues, provide hardware specification and debug.log.

closed 2020-11-09 by SergeySlice