CloverHackyColor/CloverBootloader #293: Does not recognize Mac Recovery Partition structure on USB

scerazy, 2020-11-03:
With rEFInd I can create USB GPT disk, install rEFInd to EFI partition, copy GUID folder from Mac Recovery Partition to HSF+ partition on USB & boot to it just fine (it recognizes Recovery on USB in the bottom row and boots perfectly)

It only needs showtools apple_recovery in its config

I could not get Clover to recognize such Recovery on USB

Am I missing something?

scerazy, 2020-11-21:
Nobody?

SergeySlice, 2021-01-27:
Resolved 5129.

closed 2021-01-27 by SergeySlice

scerazy, 2021-03-07:
Resolved? 5131 still the same. 

In Mojave/Catalina, one must rename copied folder (as is on Recovery partition ie. 0CFBB9AA-CC1D-4192-8072-EA1FA1797AE6) to com.apple.recovery.boot - ofcourse adjusting com.apple.Boot.plist inside it:

Code:

<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
    <key>Kernel Cache</key>
    <string>\com.apple.recovery.boot\immutablekernel</string>
    <key>Kernel Flags</key>
    <string>root-dmg=file:///com.apple.recovery.boot/BaseSystem.dmg -v</string>
</dict>
</plist>

Latest Clover does display Boot from Recovery on USB
Which means Clover has hard-coded search for com.apple.recovery.boot folder (unlike rEFInd which also recognizes UUID recovery folder)