CloverHackyColor/CloverBootloader #295: Internal HD sata/sas not appearing

ioriyagami2020, 2020-11-07:
I've installed 5126,still on catalina 10.15.7. Internal sata and sas drive is not appearing. Using disk utility still it doesnt appear.
From 5122 all was normal.

SergeySlice, 2020-11-09:
Any proof?

ioriyagami2020, 2020-11-10:
[image]

[image]

In bios it detect the internal drives. In terminal it only list the boot drive

SergeySlice, 2020-11-18:
Is your SATA controller in AHCI mode? The system will not see drives in ATA mode.

ioriyagami2020, 2020-11-18:
Sata mode is in ahci. Tried with a clean install from catalina, everything was ok, sata, usb3 everything was perfect. After big sur update, it seems to mess up the config.plist.

SergeySlice, 2020-11-28:
Do you have apfs driver installed?

ioriyagami2020, 2020-11-29:
Yes, apfs driver is loaded. Or is it because im on intel c600 chipset?

Mory5, 2020-12-04:
did you fix it i have the same issue

ioriyagami2020, 2020-12-06:
After tinkering a few days, i was able to get sata/sas & dvd recognized by clover. Somehow S/L/E was lock from admin, had to use another machine to put sata unsupport kext into that folder. Boot up and now it detect everything. What puzzles me is i cant directly alter that folder. Even after entering my admin password.

Mory5, 2020-12-06:
you but kext for your efi or what ? what kext you but

ioriyagami2020, 2020-12-07:
Sata.unsupported.kext - inside S/L/E . When inside clover kext folder it will not detect my internal sas/sata. Its weird, because in catalina everything was ok

Mory5, 2020-12-07:
not working too you use ssd or m.2 samsung ?

ioriyagami2020, 2020-12-08:
It was a nomal wd hard drive, the boot drive is Samsung ssd

SergeySlice, 2020-12-21:
There is SATA problem for Hackintoshes in BigSur. It is not Clover affair.

closed 2020-12-21 by SergeySlice