CloverHackyColor/CloverBootloader #297: Not able to boot bigSurr

nikhil19rathore99, 2020-11-13:
it goes into loop, i've used 5126 and last released  kexts in clover and i am not able to pass it through 3rd restart of installation of bigSurr, i have not access to any macOs ryt now please help
ASUS X556UQK
[image]

HandaTakuya, 2020-11-13:
my lenovo L340 Gaming can't boot to install step (update from OTA) 
Clover 5126 All kext last version.

https://cdn.discordapp.com/attachments/669350354814763028/776735705615171604/IMG_20201113_160939.jpg

bangbaew, 2020-11-14:
Same, can't boot Big Sur installer (both OTA and Olarila) with Clover 5126, up to date kexts, installed OcQuirks and OcRuntime drivers, tried different Quirks like my pc that use OpenCore and able to boot Big Sur, but still no luck on Clover. Always got kernel panic message.

pagimamaun, 2020-11-14:
Same issue can't boot big sur with 5126 only black screen verbose give nothing
Big Sur working fine with opencore

Aldo97, 2020-11-15:
Have you booted from preboot? I solved booting via preboot on my Lenovo Z50-70.

nikhil19rathore99, 2020-11-16:
> Have you booted from preboot? I solved booting via preboot on my Lenovo Z50-70.

Preboot option not shows in clover for me,

Aldo97, 2020-11-16:
> > Have you booted from preboot? I solved booting via preboot on my Lenovo Z50-70.
> 
> Preboot option not shows in clover for me,  

Fn+F3

nikhil19rathore99, 2020-11-16:
> > > Have you booted from preboot? I solved booting via preboot on my Lenovo Z50-70.
> > 
> > 
> > Preboot option not shows in clover for me,
> 
> Fn+F3

Thanks it works with preboot now

bangbaew, 2020-11-16:
> Have you booted from preboot? I solved booting via preboot on my Lenovo Z50-70.

[image]

Mine booting from preboot also gives the panic message

nikhil19rathore99, 2020-11-16:
But one more problem is there sometimes it takes longer boot time because of tx_flush....
Anyone how to stop it
Thanks

bangbaew, 2020-11-16:
> > Have you booted from preboot? I solved booting via preboot on my Lenovo Z50-70.
> 
> [image]
> 
> Mine booting from preboot also gives the panic message

Update: I can boot Big Sur installer now, but I needed to block injected kexts in Clover: AirportBrcmFixup, and also some other kexts such as NoTouchID, NullEthernet, NVMeFix, VooodooI2C, VoodooPS2 etc.
[image]

Aldo97, 2020-11-16:
> Update: I can boot Big Sur installer now, but I needed to block injected kexts in Clover: AirportBrcmFixup, and also some other kexts such as NoTouchID, NullEthernet, NVMeFix, VooodooI2C, VoodooPS2 [...]

Easy to solve (I think). 
Do you removed, in AirportBrcmFixup kext, AirPort_Brcm4360 ? 
Then whats the version you use of voodoops2?
Don't use the last version of PS2 with voodooi2c, use version 2.0.4 of PS2 and let us know.
I hope this helped you.

bangbaew, 2020-11-16:
> Easy to solve (I think).
> Do you removed, in AirportBrcmFixup kext, AirPort_Brcm4360 ?
> Then whats the version you use of voodoops2?
> Don't use the last version of PS2 with voodooi2c, use version 2.0.4 [...]

I didn't remove the kexts from the EFI because I don't know which kexts caused kernel panic, I just blocked injected kexts from Clover options, but AirportBrcmFixup is the one that caused PXSX stall loop.

Aldo97, 2020-11-16:
> I didn't remove the kexts from the EFI because I don't know which kexts caused kernel panic, I just blocked injected kexts from Clover options, but AirportBrcmFixup is the one that caused PXSX stall [...]

You need to remove only the plug-in AirPort_Brcm4360.kext in AirportBrcmFixup.
Big sur cut brcm 4360 support so it fails to boot with this plugin (I had the same problem)

nikhil19rathore99, 2020-11-16:
[image]
it stucks here for 2-3 mins before goes to login screen.

Aldo97, 2020-11-16:
> [image]
> it stucks here for 2-3 mins before goes to login screen, anybody please suggest fix for that.

Uhm. It's like the error I have.
In my case it's ACPI_SMC_PLATFORMPLUGIN that is stalling... If you solve let me know, don't know if it can be related, but just in case...

Aldo97, 2020-11-16:
> > > > Have you booted from preboot? I solved booting via preboot on my Lenovo Z50-70.
> > > 
> > > 
> > > Preboot option not shows in clover for me,
> > 
> > Fn+F3
> 
> Thanks it works with preboot now

You can't see preboot from clover because you have hidden it in the config.plist.
Please, modify your config plist, finding "hide", and comment preboot or replace it with the name of your drive.
You can boot big sur only from preboot, so I suggest you to hide the other boot option. You can hide that boot option replacing "preboot" with the drive name in the "hide" field.

bangbaew, 2020-11-16:
> Easy to solve (I think).
> Do you removed, in AirportBrcmFixup kext, AirPort_Brcm4360 ?
> Then whats the version you use of voodoops2?
> Don't use the last version of PS2 with voodooi2c, use version 2.0.4 [...]

I've tried VoodooPS2Controller 2.0.4 and it boots really fast, my touchscreen works perfectly, thanks a lot.

bangbaew, 2020-11-16:
Now the problem is that I can't install app from app store, it stucks at 100%
[image]

bangbaew, 2020-11-16:
> Easy to solve (I think).
> Do you removed, in AirportBrcmFixup kext, AirPort_Brcm4360 ?
> Then whats the version you use of voodoops2?
> Don't use the last version of PS2 with voodooi2c, use version 2.0.4 [...]

Now, my problem is that if I use VoodooPS2 and I2C, I won't be able to use AppleALC and I can't mount EFI, the Windows partition will disappear, but if I block VoodooPS2 and I2C, the problem's gone, I can use AppleALC and mouse EFI. I don't know why this happens.

Aldo97, 2020-11-16:
> Now, my problem is that if I use VoodooPS2 and I2C, I won't be able to use AppleALC and I can't mount EFI, the Windows partition will disappear, but if I block VoodooPS2 and I2C, the problem's gone, [...]

Yeah, this happens also to me.
I haven't tried to boot without voodooi2c and voodoops2 but I can't mount the Efi partition, AppleALC doesn't work and I have a slow boot.
The funny thing is that after some reboots occasionally the Efi partition can be mounted, the boot is as it was on Catalina and AppleALC works only for the mic and the headphones (the speakers doesn't work even in this case...)
I should try to boot without voodooi2c and with the last version of voodoops2.
It's possibile that voodooi2c needs to be updated to support big sur (an update is however needed because the incompatibility with the last version of voodoops2 due to voodooinput).
I can't try now, can you try to boot with the latest version of voodoops2 and without voodooi2c?
Let me now if you still have problems. If not I think we need to open an issue report in the voodooi2c github page.

Aldo97, 2020-11-16:
@bangbaew I tried what I said.
I can confirm that the slow boot, the Efi mount problem and AppleALC not working all the reboot is due to Voodooi2c or VoodooPS2 2.0.4.
I disabled voodooi2c and used voodoops2 2.1.8 and I solved all but one problem.
The speaker doesn't work. It can be because I made my own fork of AppleALC (even if in Catalina it worked).
I need to investigate the problem.
I have no time now to open an issue report on voodooi2c but later today I will open one.
Let me know if you solve doing the same.

bangbaew, 2020-11-16:
> @bangbaew I tried what I said.
> I can confirm that the slow boot, the Efi mount problem and AppleALC not working all the reboot is due to Voodooi2c or VoodooPS2 2.0.4.
> I disabled voodooi2c and used [...]

Yes, I'm able to boot with latest VoodooPS2 fastly without VoodooI2C enabled,
If I enable VoodooI2C, I'll get stuck at x86PlatformPlugin.
May I ask if you can download apps from App Store normally? I can't download anything, it stucks at 100%.

Aldo97, 2020-11-16:
> > @bangbaew I tried what I said.
> > I can confirm that the slow boot, the Efi mount problem and AppleALC not working all the reboot is due to Voodooi2c or VoodooPS2 2.0.4.
> > I disabled voodooi2c and used voodoops2 2.1.8 and I solved all but one problem.
> > The speaker doesn't work. It can be because I made my own fork of AppleALC (even if in Catalina it worked).
> > I need to investigate the problem.
> > I have no time now to open an issue report on voodooi2c but later today I will open one.
> > Let me know if you solve doing the same.
> 
> Yes, I'm able to boot with latest VoodooPS2 fastly without VoodooI2C enabled,
> If I enable VoodooI2C, I'll get stuck at x86PlatformPlugin and tx_flush.
> May I ask if you can download apps from App Store normally? I can't download anything, it stucks at 100%. 

Just installed telegram lite from the app store without any problem.
I just checked also AppleALC. It seems that the custom id I made doesn't work on big sur... Very strange... Using a standard id I have the speakers but not the mic (the mic is the reason why I made the custom id).
It's so strange as in Catalina this was not the case.

Aldo97, 2020-11-16:
I fixed also voodooi2c.
Just remove voodooinput.kext and voodooinput.kext.dSYM in the voodooi2c plug-in folder (just like AirportBrcmFixup).
Now I need only to fix AppleALC and I fixed all the problems I had.
Let me know if it fixes your problem too.

nikhil19rathore99, 2020-11-16:
> I fixed also voodooi2c.
> Just remove voodooinput.kext and voodooinput.kext.dSYM in the voodooi2c plug-in folder (just like AirportBrcmFixup).
> Now I need only to fix AppleALC and I fixed all the problems I had.
> Let me know if it fixes your problem too.
> For the app store problem... Try to do a clean install.

If voodooinput is removed then major gestures not works

Aldo97, 2020-11-16:
> If voodooinput is removed then major gestures not works

Gestures like? I have touchscreen on my device and I can use all the gestures in settings/trackpad.
I don't know if you are referring to those gestures. If yes I can guess voodooi2c simply uses voodooinput from PS2.

nikhil19rathore99, 2020-11-16:
> Gestures like? I have touchscreen on my device and I can use all the gestures in settings/trackpad.
> I don't know if you are referring to those gestures. If yes I can guess voodooi2c simply uses [...]

yes it's true but after removing voodoo input.kext my I2C trackpad gestures not working, u said I can keep VoodooPS2/plugin/voodooinput.kext instead of voodooi2c/plugin/voodooinput.kext

Aldo97, 2020-11-16:
> yes it's true but after removing voodoo input.kext my I2C trackpad gestures not working, u said I can keep VoodooPS2/plugin/voodooinput.kext instead of voodooi2c/plugin/voodooinput.kext

In fact I kept voodooinput from voodooi2c and I'm still able to use gestures on the touchscreen.
If you have problem doing what I have done you can try keeping voodooi2c voodooinput and removing it from voodoops2.
I think however that somebody needs to open an issue repot for voodooi2c

Aldo97, 2020-11-16:
I think you can close this issue.

closed 2020-11-17 by nikhil19rathore99