CloverHackyColor/CloverBootloader #299: cant build the clover

UchihaSR, 2020-11-14:
i am trying to compile the clover bootloader from source by doing sudo .\buildme  but it failed by showing following error
build.py...
/home/itachi/CloverBootloader/Clover.dsc(294): error 000E: File/directory not found in workspace
	/home/itachi/CloverBootloader/OpenCorePkg/Platform/OpenCore/OpenCoreLib.inf

SergeySlice, 2020-11-18:
You should take linked project such way
git clone --recurse-submodule ....

UchihaSR, 2020-11-20:
make: [GNUmakefile:1146: /home/itachi/CloverBootloader/Build/Clover/RELEASE_GCC53/FV/Ffs/FC5C7020-1A48-4198-9BE2-EAD5ABC8CF2FBdsDxe/BdsDxeOffset.raw] Error 1 (ignored)
/home/itachi/CloverBootloader/OpenCorePkg/Include/Acidanthera/Library/OcStorageLib.h:104:1: error: type of 'OcStorageInitFromFs' does not match original declaration [-Werror=lto-type-mismatch]
 OcStorageInitFromFs (
 ^
/home/itachi/CloverBootloader/OpenCorePkg/Library/OcStorageLib/OcStorageLib.c:215:1: note: 'OcStorageInitFromFs' was previously declared here
 OcStorageInitFromFs (
 ^
/home/itachi/CloverBootloader/OpenCorePkg/Library/OcStorageLib/OcStorageLib.c:215:1: note: code may be misoptimized unless -fno-strict-aliasing is used
lto1: all warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/gcc returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [GNUmakefile:24224: /home/itachi/CloverBootloader/Build/Clover/RELEASE_GCC53/X64/rEFIt_UEFI/refit/DEBUG/CLOVERX64.dll] Error 1

build.py...
 : error 7000: Failed to execute command
	make tbuild [/home/itachi/CloverBootloader/Build/Clover/RELEASE_GCC53/X64/rEFIt_UEFI/refit]

build.py...
 : error F002: Failed to build module
	/home/itachi/CloverBootloader/rEFIt_UEFI/refit.inf [X64, GCC53, RELEASE]

- Failed -
Build end time: 13:30:23, Nov.20 2020
Build total time: 00:00:11

SergeySlice, 2020-11-28:
resolved

closed 2020-11-28 by SergeySlice