CloverHackyColor/CloverBootloader #300: Big Sur 11.01 Clover R5126 Can't download or update from App Store

JustFou, 2020-11-14:
Hi Clover Team,

I've seen this issue on my machine and a lot of there people. It seems on Big Sur any machine running clover 5126 and Big Sur (doesn't seem to appear on OpenCore) will get this bug. Basically everything else works fine but as soon as you download something on the app store it will start but stops forever at about 80%( or done downloading but doesn't switch to install). I can provide pictures of the bug as well. Also once your restart the bugged apps stays in the launchpad as greyed out icons. No clue what's causing this but I'll report this issue to the team.

DecadentIpsum, 2020-11-15:
The same happens to me as well, BiG Sur 11.01 - Clover 5126.

HandaTakuya, 2020-11-16:
some

FindVPS, 2020-11-16:
Same problem BiG Sur 11.01 - Clover 5126. 
I try to downgrade to Clover 5125 - do not help

J0uj0u, 2020-11-16:
Same for me ..

JustFou, 2020-11-16:
If someone find a fix for this please share... at the moment the only fix is switching to OpenCore!

chriseg93, 2020-11-17:
Same issue here

xylz0928, 2020-11-17:
same issue here on GigaZ370HD3P, Dell E7490

oneszeos, 2020-11-18:
same issue

alexjjassuncao, 2020-11-18:
I'm not sure but it could be actually a issue on Big Sur.

There's a thread on Developer Forums talking about this.

👉 [Big Sur Mac App Store stuck installing app, never completes](https://developer.apple.com/forums/thread/662756)

ZaydelSenpai, 2020-11-18:
> I'm not sure but it could be actually a issue on Big Sur.
> 
> There's a thread on Developer Forums talking about this.
> 
> 👉 [Big Sur Mac App Store stuck installing app, never completes](https://developer.apple.com/forums/thread/662756)

Posted 5 minutes ago:

Can someone please try this solution from de official Apple Developer Forum?
[image]

DecadentIpsum, 2020-11-18:
> Posted 5 minutes ago:
>
> Can someone please try this solution from de official Apple Developer Forum?
> [image]

Didn't work for me :(

alexjjassuncao, 2020-11-18:
> Didn't work for me :( The updates are still pending for install

Didn't work for me also.

jianglinchun, 2020-11-19:
Same for me .....

Albert0987654321, 2020-11-19:
Same for me :(

jianglinchun, 2020-11-19:
[image]
I'm wondering whether this is reason about this problem. Big Sur is compatible both M1 & intel cpu? And the OS could not determine the cpu type?????

octoforce, 2020-11-19:
same, still exist this problem after switching to OC

oneszeos, 2020-11-20:
A new installation can solve this problem

xylz0928, 2020-11-20:
> A new installation can solve this problem

do you know if re-install by booting from a installer, cover the current disk, will it be solved?
(mac is my primary OS, i really do not want to rebuild all things)

BTW i will test this today evening

xylz0928, 2020-11-20:
> > A new installation can solve this problem
> 
> do you know if re-install by booting from a installer, cover the current disk, will it be solved?
> (mac is my primary OS, i really do not want to rebuild all things)
> 
> BTW i will test this today evening

Test success! can solve without lose your current data:

1. use full installer media (U disk or partition)
2. boot clover r5126, boot installer
3. install 11.0.1 and cover the current partition (we call this partition name as Macintosh in this example)
4. on the 2nd step of installation, choose boot install of your Macintosh partition
5. on the 3rd step of the installation, **MUST** choose Boot **preboot** from Macintosh
6. then you can success install/ update appa from mac app store

the key point is 'we haven't finished the installation of the installation of macOS then we face that issue'

ZaydelSenpai, 2020-11-20:
> Test success! can solve without lose your current data:
>
> 1. use full installer media (U disk or partition)
> 2. boot clover r5126, boot installer
> 3. install 11.0.1 and cover the current partition (we [...]

So, Basically do you reinstalled on the same drive where do you upgraded from Catalina? That's it? :O

xylz0928, 2020-11-20:
> So, Basically do you reinstalled on the same drive where do you upgraded from Catalina? That's it? :O

i upgrade to Big Sur from Catalina, then i faced that issue.
i reinstalled Big Sur on the same drive.
Boot via preboot, solved

ZaydelSenpai, 2020-11-20:
> i upgrade to Big Sur from Catalina, then i faced that issue.
> i reinstalled Big Sur on the same drive.
> Boot via preboot, solved

I don't get the part of "Reinstalled big sur" do you just selected again the Macintosh partition without formating it first? Like, Overwritting it?

csoto09, 2020-11-20:
> I don't get the part of "Reinstalled big sur" do you just selected again the Macintosh partition without formating it first? Like, Overwritting it?

Precisely. I made a USB installer and booted to it following the steps listed above by @xylz0928 and chose my existing Mac partition without formatting. Ran without any issues and the App Store issue is now solved for me.

JustFou, 2020-11-20:
> Test success! can solve without lose your current data:
>
> 1. use full installer media (U disk or partition)
> 2. boot clover r5126, boot installer
> 3. install 11.0.1 and cover the current partition (we [...]

This works actually. I messed up somewhere because now I can only boot from preboot lol oops! But everything works perfectly so it's a non issue for now. Let's see in the future!

DecadentIpsum, 2020-11-20:
> Precisely. I made a USB installer and booted to it following the steps listed above by @xylz0928 and chose my existing Mac partition without formatting. Ran without any issues and the App Store [...]

Actually the step I am not getting is to boot from preboot, will I have always to boot from there? I will appreciate a short explanation!

xylz0928, 2020-11-21:
> Actually the step I am not getting is to boot from preboot, will I have always to boot from there? I will appreciate a short explanation!

  yes sure we need to boot from preboot always, i also tested on OC, it's the same, because from Big Sur, the system partition has been made as a snapshot, (to protect system files i guess). (u can c in diskutil)
   but there is some solutions to cancel the  snapshot and back to directly use the partition.(not regarding to the bootloader no matter clover or oc). if you prefer u can have a search.
    but for me i prefer to follow the default of macOS.

DecadentIpsum, 2020-11-21:
> > > > > > > > > A new installation can solve this problem
> > 
> > 
> > > > > > > >
> > 
> > 
> > > > > > > >
> > 
> > 
> > > > > > > > do you know if re-install by booting from a installer, cover the current disk, will it be solved?
> > 
> > 
> > > > > > > > (mac is my primary OS, i really do not want to rebuild all things)
> > 
> > 
> > > > > > > > BTW i will test this today evening
> > 
> > 
> > > > > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > > > > > Test success! can solve without lose your current data:
> > 
> > 
> > > > > > >
> > 
> > 
> > > > > > > 1. use full installer media (U disk or partition)
> > 
> > 
> > > > > > > 1. boot clover r5126, boot installer
> > 
> > 
> > > > > > > 1. install 11.0.1 and cover the current partition (we call this partition name as Macintosh in this example)
> > 
> > 
> > > > > > > 1. on the 2nd step of installation, choose boot install of your Macintosh partition
> > 
> > 
> > > > > > > 1. on the 3rd step of the installation, **MUST** choose Boot **preboot** from Macintosh
> > 
> > 
> > > > > > > 1. then you can success install/ update appa from mac app store
> > 
> > 
> > > > > > >
> > 
> > 
> > > > > > > the key point is 'we haven't finished the installation of the installation of macOS then we face that issue'
> > 
> > 
> > > > > >
> > 
> > 
> > > > > >
> > 
> > 
> > > > > > So, Basically do you reinstalled on the same drive where do you upgraded from Catalina? That's it? :O
> > 
> > 
> > > > >
> > 
> > 
> > > > >
> > 
> > 
> > > > > i upgrade to Big Sur from Catalina, then i faced that issue.
> > 
> > 
> > > > > i reinstalled Big Sur on the same drive.
> > 
> > 
> > > > > Boot via preboot, solved
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > I don't get the part of "Reinstalled big sur" do you just selected again the Macintosh partition without formating it first? Like, Overwritting it?
> > 
> > 
> > > 
> > 
> > 
> > > Precisely. I made a USB installer and booted to it following the steps listed above by @xylz0928 and chose my existing Mac partition without formatting. Ran without any issues and the App Store issue is now solved for me.
> > 
> > 
> > Actually the step I am not getting is to boot from preboot, will I have always to boot from there? I will appreciate a short explanation!
> 
> yes sure we need to boot from preboot always, i also tested on OC, it's the same, because from Big Sur, the system partition has been made as a snapshot, (to protect system files i guess). (u can c in diskutil)
> but there is some solutions to cancel the snapshot and back to directly use the partition.(not regarding to the bootloader no matter clover or oc). if you prefer u can have a search.
> but for me i prefer to follow the default of macOS.

Thanks a lot for the help!

Hoangpham89, 2020-11-21:
Hi everybody. I'm upgrading directly from Catalina 10.15.7 to Big Sur used the latest clover r5126 and the latest kexts but I've just stuck as the image below, does anybody knows how to fix this case? please help me, thank you very much.
[image]

J0uj0u, 2020-11-21:
> Hi everybody. I'm upgrading directly from Catalina 10.15.7 to Big Sur used the latest clover r5126 and the latest kexts but I've just stuck as the image below, does anybody knows how to fix this case? please help me, thank you very much.
> [image]

Try to run the preboot drive of installation. If it's not visible use Fn + F3 or F3 to see hidden drive.

Hoangpham89, 2020-11-21:
> Try to run the preboot drive of installation. If it's not visible use Fn + F3 or F3 to see hidden drive.

I've just tried as your solution but could not solve in this case.haizz

alexjjassuncao, 2020-11-23:
> Test success! can solve without lose your current data:
>
> 1. use full installer media (U disk or partition)
> 2. boot clover r5126, boot installer
> 3. install 11.0.1 and cover the current partition (we [...]

It worked, thanks! 😊

Albert0987654321, 2020-11-25:
> It worked, thanks! 😊

Worked for me too! 🙌🏻

HandaTakuya, 2020-11-26:
> Test success! can solve without lose your current data:
>
> 1. use full installer media (U disk or partition)
> 2. boot clover r5126, boot installer
> 3. install 11.0.1 and cover the current partition (we [...]

Yeahhh !! IS WORK !!

But after step 3 boot to Preboot I can't boot form Macintosh Partition my Hackintosh is panic !!

[image]

FindVPS, 2020-11-26:
Somebody try after update on clover r5127?

DecadentIpsum, 2020-11-26:
> Somebody try after update on clover r5127?

Same issue after updating on Clover r5127.

jianglinchun, 2020-11-27:
> > Somebody try after update on clover r5127?
> 
> The issue still exists after updating on Clover r5127.

I had upgrade to r5127 and confirmed that the problem still exists.

anhvu-sg, 2020-11-27:
> > Somebody try after update on clover r5127?
> 
> The issue still exists after updating on Clover r5127.

I also have a problem with r5126 so this morning I tried r5127.
My specs:
- z390 i pro wifi
- i9 9900k
- VGA: Radeon rx590
Here are all the steps:
1. My current Clover version is r5121
2. Download the r5127 and install
3. Update Quirks by following this [post](https://www.tonymacx86.com/threads/new-clover-release-v5-1-r5123-4-5-6-7-big-sur-support.304543/page-4#post-2180221)
4.  Reboot
5. I can boot to Catalina (in r5126 I cannot boot)
6. Upgrade to Big Sur via Appstore
7. After 30 mins, it's success

Problem:
- USB does not work, I will try to check with the latest kexts tonight and share with you the information

anhvu-sg, 2020-11-28:
> > > Somebody try after update on clover r5127?
> > 
> > 
> > The issue still exists after updating on Clover r5127.
> 
> I also have a problem with r5126 so this morning I tried r5127.
> My specs:
> 
> * z390 i pro wifi
> * i9 9900k
> * VGA: Radeon rx590
>   Here are all the steps:
> 
> 1. My current Clover version is r5121
> 2. Download the r5127 and install
> 3. Update Quirks by following this [post](https://www.tonymacx86.com/threads/new-clover-release-v5-1-r5123-4-5-6-7-big-sur-support.304543/page-4#post-2180221)
> 4. Reboot
> 5. I can boot to Catalina (in r5126 I cannot boot)
> 6. Upgrade to Big Sur via Appstore
> 7. after reboot, I choose to install Big Sur from preboot
> 8. After 30 mins, it's success
> 
> Problem:
> 
> * USB does not work, I will try to check with the latest kexts tonight and share with you the information

I confirmed it works perfectly with r5127, thanks to the whole team 👍

GoodBoris, 2020-12-02:
Well done. BS is installed via 'covdr' way and preboot option. How to get rid of that preboot option and start hitting to OSx from the main partition as usual?

SergeySlice, 2020-12-09:
Always use Preboot and be happy!

closed 2020-12-09 by SergeySlice

GoodBoris, 2020-12-09:
> Always use Preboot and be happy!

How to make it as default option? To not press F3 -> select preboot each time the PC is booting?

alessaba, 2020-12-09:
yeah that would make people happier, if for BigSur Preboot was detected automatically as the default option

vaczi001, 2021-01-15:
don't hide Preboot, find Preboot's UUID then but it to GUI/DefaultBootVolume, then it'll automatically boot. 
You can even create custom entry: 
[Create Custom Entries to Hide Preboot and Recovery APFS Volumes, Fusion Drive Installation with Clover Configurator](https://www.insanelymac.com/forum/topic/326527-create-custom-entries-to-hide-preboot-and-recovery-apfs-volumes-fusion-drive-installation-with-clover-configurator/?do=findComment&comment=2743076)

retrodude, 2021-01-16:
I have the same Problem.  Using Clover 5128.

I can't get the suggested Over-Install to work,
The description is confusing to me.
I suppose I choose the wrong Partitions... ?

If I press F3, I can choose the following Partitions:
1. - Boot UEFI Internal from EFI
2. - Boot Mac OS X Install for System - Daten via Preboot
3. - Boot Mac OS X from System - Daten via Preboot
4. - Boot macOS from System - Daten
5. - Boot Mac OS X Recovery for System - Daten via Recovery
6. - Boot macOS from System

My BigSur Installation (installed over Mojave) with the AppStore Problem is on the Partition named "System".
What I've done now is:
- Started the BigSur Installer from my running System-Partion (6) and selected to install again on "System"
- At the first reboot (yours step2) I selected Partition 2 - and the Installer works for about 15 Minutes
- at the second reboot (yours step3) I selected Partition 3.  After about 20 seconds booting the screen gets black and nothing happens anymore.  :-(

What am I doing wrong?
Can still boot from 6 (seems unchanged).

vaczi001, 2021-01-16:
@retrodude 

1. Don't hide Preboot in config.plist
2. Boot Mac OS X Install via Preboot
3. Choose that partition every time until it disappears
4. After disappeared use Boot Mac OS X from System via Preboot
5. After installation is finished boot ONLY from Preboot
If you mess up you'll need to start the installation again

If you want, follow this to make Custom Entries for Preboot or boot it automatically (if you have 1 Preboot the just write "Preboot" to DefaultBootVolume) [comment 760890340](https://github.com/CloverHackyColor/CloverBootloader/issues/300#issuecomment-760890340)

retrodude, 2021-01-16:
@vaczi001 
Thank you very much!!!
Your explaination was clear and AppStore now works also for me!

But hiding partitions in Clover is a real nightmare of user interface and logic. :-(
Tried it many hours and reading a lot of posts to this but without success. Give up now.
Hopefully I will never select accidentely "Boot macOS from System"...  (which I did not get to hide).

vaczi001, 2021-01-17:
to hide that simply write "macOS" (system drive's name) to Hide

retrodude, 2021-01-17:
This has no Effect. Both entries are still there. 
Why do you suggest "macOS" - theres is no Volume with this name?
If I add "System" to Hide, then both entries disapear. :-( Why???

My diskutil list:
```
/dev/disk3 (synthesized):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      APFS Container Scheme -                      +499.9 GB   disk3
                                 Physical Store disk0s2
   1:                APFS Volume ⁨System - Daten⁩          193.0 GB   disk3s1
   2:                APFS Volume ⁨Preboot⁩                 292.2 MB   disk3s2
   3:                APFS Volume ⁨Recovery⁩                613.6 MB   disk3s3
   4:                APFS Volume ⁨VM⁩                      1.1 MB     disk3s4
   5:                APFS Volume ⁨System⁩                  15.0 GB    disk3s5
   6:              APFS Snapshot ⁨com.apple.os.update-...⁩ 15.0 GB    disk3s5s1

```

Adding  the GUID-String (taken from boot-log) to Hide also has no effect:
HD(2,GPT,C0BA25F2-A0F9-43E4-A24B-7F12CE5015A0) 
Just like Volume/PartitionUUID (taken from diskutil info) as String:
748606DD-D642-4E7E-9A56-2041F9B001A6
Interesting that hiding "Recovery" with putting this to Hide works as expected instead...

LeeBinder, 2021-01-27:
Anybody here managed to

1. create a GUI custom entry for Big Sur Preboot labeled "Big Sur" or alike so it does not always appear as "Mac OS X from Big Sur via Preboot" in Clover boot screen?
2. hide the Big Sur system entry?

Still not working here neither with Clover 5128 nor with latest Clover nightly...

My system info: macOS installs and volume labels: Big Sur, Catalina, Mojave

Big Sur Preboot:
APFS volume name/ label: Preboot

Volume: PciRoot(0x0)\Pci(0x17,0x0)\Sata(0x1,0xFFFF,0x0)\HD(6,GPT,233EA3E0-9DBD-49F2-8F51-DA57BCC54B87,0x8CF0000,0xDACB000)\VenMedia(BE74FCF7-0B7C-49F3-9147-01F4042E6842,7BE84A7CA39E5F4C8755EACFB18751E5)

UUID: 233EA3E0-9DBD-49F2-8F51-DA57BCC54B87

APFS File System UUID: 7C4AE87B-9EA3-4C5F-8755-EACFB18751E5

APFS Container UUID: 2466E13F-E4BB-4B37-BD4D-DCD0E74E83CE, ApfsRole=0x10

APFS target UUID: BA2910E1-CEA3-4140-88E3-89C70BCF433E

Would you please share your GUI section from Clover config.plist if you have a working custom entry. Thanks.

vaczi001, 2021-01-27:
@LeeBinder 

config.plist:
[image]

Volume UUID: diskutil list, Container disk1 -> disk0s2
diskutil info disk0s2
[image]

Preboot ID and boot.efi location: 
[image]

After that you'll see "macOS Big Sur" entry instead Preboot. 

To hide the "main" partition write "your macOS drive name" or it's UUID to Hide.

LeeBinder, 2021-01-27:
@vaczi001 thank you very much, but.. even though I followed everything meticulously according to your fine instructions incl. the backslashes, double/ triple checked.. it's not working here. I do get the icon in the GUI, but then -> kernel panic.

I also just updated to the brand-new Clover 5129 final - same thing.

Can you do me two favors:
1. in Clover GUI, right click onto your `macOS Big Sur` entry, make a photo/ screenshot of that screen, and attach it here
2. post your output from hackintool -> logs: Boot

either in a spoiler

```
<details>
  <summary>click to expand</summary> 

  Spoiler text. Note that it's important to have a SPACE after the summary tag. You should be able to write any markdown you want inside the `<details>` tag... just make sure you close `<details>` afterward.
</details>
```
or attached as txt file.

That would be awesome. Thanks for your cooperation!

PS: can you still boot into Big Sur via your custom entry after the Clover 5129 update?

vaczi001, 2021-01-27:
just to not bother this issue, I replied on Insanelymac: https://www.insanelymac.com/forum/topic/346498-gui-custom-entry-for-big-sur-preboot-how-to/?do=findComment&comment=2749567

maycon2-0, 2021-04-17:
I have the same problem and I fix after discover the NullEthernet stop to work on Big Sur, now I have one network adpater ho work on Big Sur and reset my BSD on Wifi to EN0 like in this video https://www.youtube.com/watch?v=IYismHReQ8A&t=157s and after restart the App Store back to work, forgive my english I`m a brazilian.

DecadentIpsum, 2021-04-17:
> I have the same problem and I fix after discover the NullEthernet stop to work on Big Sur, now I have one network adpater ho work on Big Sur and reset my BSD on Wifi to EN0 like in this video [...]

Hey @maycon2-0 did you follow all the steps in the video in order to make the app store to work?

maycon2-0, 2021-04-17:
> Hey @maycon2-0 did you follow all the steps in the video in order to make the app store to work?

Only the first, deleting the network preferences to reset de adapter to EN0, the rest of the video is only needed if you have problems with iMessage.
See the video from 0:00 until 1:56 that’s the solution for me.

spsharma2018, 2021-05-19:
Hello,
Nice solution. How do I make my "Custom Entry" as the default boot (timeout=5 sec) at the clover boot screen ?

kushwavez, 2021-05-19:
@spsharma2018
If I remember correctly I could do that by inputting the UUID as the DefaultVolume name.

spsharma2018, 2021-05-19:
Thanks. It worked.