CloverHackyColor/CloverBootloader #305: Serial number not read with 5126

alainjroy, 2020-11-17:
When switching from 5122 to 5126, after having added quirks for the dell E7270, the system is unable to read the serial number and model. What I'm I missing ?

ryulove3, 2020-11-17:
RtVariables-->CsrActiveConfig -->0x3E7
SystemParameters --->InjectSystemID -->yes

augustozanellato, 2020-11-17:
> RtVariables-->CsrActiveConfig -->0x3E7
> SystemParameters --->InjectSystemID -->yes

I’m also on a Dell laptop, I think the issue is somewhere else, when booting Dell machines with OpenCore you need to enable Kernel > Quirks > CustomSMBIOSGuid and set PlatformInfo > UpdateSMBIOSMode to Custom, currently there isn’t a way to do that when using Clover, I think some switches should be added to Clover config.plist

alainjroy, 2020-11-18:
I tried ryulove3 suggestion to no avail. I already had 
SystemParameters --->InjectSystemID -->yes
and changed RtVariables -- CsrActiveConfig from 0x67 to ox3E7.
Now I'm trying to find a way to tell clover to enable Kernel > Quirks > CustomSMBIOSGuid and set PlatformInfo > UpdateSMBIOSMode to Custom.

ryulove3, 2020-11-18:
Hp 850G1(clover r5126) , Dell E6540 (Clover r5126) .. no problems about Apple ID
..If you use 'VirtualSMC', try to ''FakeSMC''.
Try and compare both. Check out nvram with HackinTool

ryulove3, 2020-11-18:
Try fakeSMC and vituralSMC, compare them. 
Check out nvram with HackinTool
Don't use CustomUUID 
'model, serialNumber,BoardSerialNumber, sistemUUID, Rom' must be fixed value.
After always check with iMessageDebug, login appleID

--------------------
Clover r5126, big sur
HP 850G1, Dell E6540 no problems (in my case)

kumakumalina, 2020-11-18:
tick quirks if necessary, leave everything default. And try to DONT tick DellSMBiosPatch
i dont know what from clover version, if you tick the DellSmbiosPatch, Clover will inject your Dell OEM info to macOS

augustozanellato, 2020-11-18:
> tick quirks if necessary, leave everything default. And try to DONT tick DellSMBiosPatch
> i dont know what from clover version, if you tick the DellSmbiosPatch, Clover will inject your Dell OEM info to macOS

Has DellSmbiosPatch behavior changed with latest clover updates? It was neeeded on Dell laptop in order to fix a truncation issue on values read from SMBIOS (see [here](https://sourceforge.net/p/cloverefiboot/tickets/203/)), it shouldn’t inject Dell OEM info

alainjroy, 2020-11-18:
Like **kumakumalina** suggested, it's the DellSmbiosPatch that causes the problem with clover 5126. By unselecting it, the system now sees the serial and other informations as expected. Thanks a lot kumakumakina !

jief666, 2020-11-18:
DellSMBIOSPatch should now work as before.
Fix in last commit : 
[CloverX64-f28e8a2-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5562993/CloverX64-f28e8a2-jief.zip)

closed 2020-11-18 by jief666