CloverHackyColor/CloverBootloader #309: r5126 stuck at [EB  LOG:EXITBS:START]

Happymax, 2020-11-19:
Hi!
I've just upgraded from Clover r5119 to r5126, added OcQuirks.efi and OpenRuntimes.efi, removed AptioMemoryFix, and added certain quirks to config.plist. Now I'm stuck at [EB  LOG:EXITBS:START] and cannot boot macOS (10.15.7). Everything else remained the same and it worked fine under r5119. I've tried changing the CFG lock related quirks, but I remembered that I already patched the BIOS long time ago, and enabling did nothing. I've attached my config and debug log. 
[debug.log](https://github.com/CloverHackyColor/CloverBootloader/files/5568788/debug.log)

[config - copy.txt](https://github.com/CloverHackyColor/CloverBootloader/files/5568797/config.-.copy.txt)

WhitemanE0, 2020-11-21:
Same issue. Kindly ask for engineering build or so

theerha2, 2020-11-23:
same issue

darrenkhouston, 2020-11-23:
Same issue for me.

[image]

WhitemanE0, 2020-11-23:
Dear Comrades,you've all come to this page becouse the same issue. Clover r5126 is nessesary to install MacOS Big Sure. But it's no working for many of us. If you've come here don't just close the page. Uppend to the this issue. This will probably motivate the our dear developers. Please register git hub to add a comment, if not yet registered.
Dear deverlopers, we all badly want to upgrade to big sure.

benwilkerson, 2020-11-24:
Same Issue for me as well.

yanlusu, 2020-11-25:
below is my quirks config section, and I can boot Catalina with 5126, hope help to you.
```xml
	<key>Quirks</key>
	<dict>
		<key>ProvideMaxSlide</key>
		<integer>0</integer>
		<key>EnableSafeModeSlide</key>
		<true/>
		<key>DisableRtcChecksum</key>
		<true/>
		<key>DisableIoMapper</key>
		<true/>
		<key>AvoidRuntimeDefrag</key>
		<true/>
		<key>ProvideConsoleGopEnable</key>
		<true/>
		<key>DevirtualiseMmio</key>
		<true/>
		<key>SyncRuntimePermissions</key>
		<true/>
		<key>ProvideCustomSlide</key>
		<true/>
		<key>ProtectUefiServices</key>
		<true/>
		<key>IncreasePciBarSize</key>
		<true/>
		<key>PowerTimeoutKernelPanic</key>
		<true/>
		<key>XhciPortLimit</key>
		<true/>
		<key>DisableLinkeditJettison</key>
		<true/>
		<key>RebuildAppleMemoryMap</key>
		<true/>
		<key>FuzzyMatch</key>
		<true/>
	</dict>
```

Fahim732, 2020-11-27:
Exactly same issue with my rig. Using i5-6500 with ASUS H110M-C mainboard and onboard graphics.

pchong90, 2020-12-03:
same issue for me with r5127. please help. I have i5-5500u

peterkkk366, 2020-12-07:
Disable CFG Lock in Bios, if no this option, check Quirks to be set to Yes: AppleCpuPmCfgLock and AppleXcpmCfgLock.

darrenkhouston, 2020-12-07:
CFG was not an issue for me. It was a conflict of Quirks and ACPI/DSDT patches that I was using... I was using this to patch the BIOS conflict for MATs. Not sure if anyone else has this experience. Since my Motherboard is using an older BIOS that doesn't support MATs, I had to use these settings.. This solved my issue, it may not work for you, but once I did all the other changes to remove APTIO memory fix and enabled all of the Open Core drivers, this allowed it to boot no issues.

EnableWriteUnprotector - True
RebuildAppleMemoryMap - False
SyncRuntimePermissions - False

If you have a motherboard who's BIOS does support MATs, usually 2018 and newer BIOS, then the opposite may apply.

EnableWriteUnprotector - False
RebuildAppleMemoryMap - True
SyncRuntimePermissions - True

riversky, 2020-12-10:
> This is most likely a Kernel Issue. CFG was not an issue for me, since it is disabled in the BIOS. It was a conflict of Quirks and ACPI/DSDT patches that I was using... I was using this to patch the BIOS conflict for MATs. Not sure if anyone else has this experience. Since my Motherboard is using an older BIOS that doesn't support MATs, I had to use these settings.. This solved my issue, it may not work for you, but once I did all the other changes to remove APTIO memory fix and enabled all of the Open Core drivers, this allowed it to boot no issues.
> 
> EnableWriteUnprotector - True
> RebuildAppleMemoryMap - False
> SyncRuntimePermissions - False
> 
> If you have a motherboard who's BIOS does support MATs, usually 2018 and newer BIOS, then the opposite may apply.
> 
> EnableWriteUnprotector - False
> RebuildAppleMemoryMap - True
> SyncRuntimePermissions - True

It works for me, thanks!

franky920920, 2020-12-21:
r5127 still stuck at [EB LOG:EXITBS:START], non of the above work for me.

mbucc9000, 2020-12-24:
Same here. Still nothing

prinze77, 2021-01-05:
Same here. Still nothing

dateno1, 2021-01-14:
I was success to boot catalina/big sur with lastest version on B75/H110/B150/Z390

5119 is old type bootloader but 5126 is OpenCore version

You will need to set quirks and remove All of aptio drivers

Also almost boards can't boot big sur with FakeSMC (Please Use FakeSMC3 or VirtualSMC)

closed 2021-01-15 by SergeySlice

wolframus, 2021-02-08:
Have the same issue. My MacOS X with big sur on board has been working fine for a couple of months now but just a couple of days ago it stopped running with this error:
[image]

freebughao, 2021-03-09:
https://github.com/CloverHackyColor/CloverBootloader/issues/309#issuecomment-774952120
Same here. Still nothing

scerazy, 2021-03-14:
On Optiplex 3046 (using OC) I needed to change DVMT Pre-alloc to 128MB

setup_var 0x350 0x4

After that error went away. Maybe worth checking on your rig? Like this?

OmkoBass, 2021-04-10:
Check your basic.sh. I opened it through virt-manager with 4096Mb of ram and 6 cores on CPU and got this message, when i lowered it back to how it is in the basic.sh file it all worked out.

Kamushy, 2022-07-10:
> RebuildAppleMemoryMap - Tru

Found a fix, Im stuck with same issue

mbucc9000, 2022-07-10:
I don't think I actually did unfortunately...