CloverHackyColor/CloverBootloader #310: r5126 does not auto-detect ESXI 6.5 volume, F3 reveals volume, boots ok

mjur3191, 2020-11-24:
Dell R820 Server with internal SD card slot 1 - running R5126 Clover Bootloader 
Additional Driver loaded: NVME 

Does not autodetect ESXI 6.5 U3 partition. When pressing F3 (show hidden) it will appear. Boots normally. 
Would like to boot without keyboard intervention. Currently have to press F3 and then select volume and press enter. 
Running Windows 10 workstation.

SergeySlice, 2020-11-25:
Do you have any Hide in your config.plist?

mjur3191, 2020-11-26:
> Do you have any Hide in your config.plist?

Default files only. Nothing has been modified. Only the NVME driver was copied from the OFF folder. All of the other files are the same as they were installed. 

Volume is HIDDEN by default - I am sure there is many ways of fixing this issue. I  do not know where the instruction manual is located.

SergeySlice, 2020-11-28:
Provide please preboot.log

mjur3191, 2020-12-09:
I would need instructions on how to create this file. I didnt see anything like this on my flash storage.

SergeySlice, 2020-12-09:
Press F2 in the Clover GUI.
Then look for the file /EFI/CLOVER/misc/preboot.log

mjur3191, 2020-12-09:
Thank you. Here is the log flile 
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/5664708/preboot.log)

SergeySlice, 2021-02-06:
Test latest commit.

SergeySlice, 2021-03-21:
looks like resolved

closed 2021-03-21 by SergeySlice

rojarrolla, 2021-08-07:
Hi, I have the same thing, my NVME is not hidden, I can only boot if I press "F3". How can I have Clover to show it to me? Thanks!

zikuan-wang, 2023-02-13:
> Hi, I have the same thing, my NVME is not hidden, I can only boot if I press "F3". How can I have Clover to show it to me? Thanks!

same issue. clover ver : 5151

LYAN0526, 2023-06-04:
same issue. clover ver : 5152