CloverHackyColor/CloverBootloader #311: Gigabyte GA-H61M-DS2 reset bios on reboot

hectorgrom, 2020-11-26:
Gigabyte GA-H61M-DS2
I5 3330
GTX650 2gb

Apparently the current version is not able to run the patch "fixRTC" or "AppleRTC", because when the PC is restarted, the bios resets

Thanks..

Great job....   :)

peterkkk366, 2020-12-07:
Maybe driver issue, if you are using HfsPlus.efi, try to use HfsPlusLegacy.efi to instead.

SergeySlice, 2020-12-09:
FixRTC or AppleRTC? What patch did you use?

hectorgrom, 2020-12-09:
Hi ... I always used "FixRTC" ... but started to reset the bios in the new versions of Clover (5120+), I tried to use "AppleRTC", but also reset the bios ... I tried to use both together ... unsuccessful ... I removed the old drivers from the "BIOS" folder ... unsuccessful ... deleted EFI and reinstalled Clover ... unsuccessful ... I'm currently using "RTCMemoryFixup.kext" + Clover 5127. .. and apparently solved it ....

closed 2021-01-02 by SergeySlice