CloverHackyColor/CloverBootloader #313: Big Sur Boot issue when not selecting Preboot

Niresh12495, 2020-11-27:
When i boot Big Sur from the partition "Big Sur" it crashes with kernel panic. and I cant see like kexts are not being injected correctly.
It works when i boot from Preboot "Big Sur"?

Is there anything am i missing on Config?

JoyZgq, 2020-11-28:
it works well by booting "via Preboot",and I've upgraded to macos 11.1 beta. Try F3 if you didn't find recovery

kumakumalina, 2020-11-28:
currently most everyone who's still using Clover only can boot to BS with Preboot entry. "Big Sur partition" seems not worked now (the issues you said, problem with kext loading). It happened since 5126

SergeySlice, 2020-11-29:
OpenCore always boot from Preboot saying nothing about this.

Always boot from Preboot!

closed 2020-11-29 by SergeySlice