CloverHackyColor/CloverBootloader #314: How to Inject AMD OpenCore Patches?

Niresh12495, 2020-11-27:
https://github.com/AMD-OSX/AMD_Vanilla

Please provide a simple sample config, i can fill the rest?

SergeySlice, 2020-11-29:
The common rule 
~~~~
OpenCore              Clover
Base                  Procedure
Comment               Comment
Count                 Count
Enabled               Disabled
Skip                  Skip
Identifier            Name
Find                  Find
Mask                  MaskFind
Replace               Replace
ReplaceMask           MaskReplace
MaxKernel             -
MinKernel             -
-                     MatchOS
-                     InfoPlistPatch
-                     RangeFind
~~~~

Niresh12495, 2020-12-02:
OpenCore  Clover
Enabled  Disabled

Is this correct?

iCanaro, 2020-12-03:
here 
https://www.macos86.it/topic/3647-clover-quirks/
https://www.macos86.it/topic/3610-new-clover-on-amd-platform/

SergeySlice, 2020-12-09:
> OpenCore Clover
> Enabled Disabled
> 
> Is this correct?

OpenCore: Enabled=YES
corresponds to 
Clover: Disabled=NO

closed 2020-12-20 by SergeySlice