CloverHackyColor/CloverBootloader #315: How to Make Preboot as Default Selection in Clover

Niresh12495, 2020-11-29:
Is there any special option to make boot from Preboot as default?
I have “Mac” as Default partition to boot from, How to use Preboot of Mac as Default?

GoodBoris, 2020-12-02:
I would say, how to move 'preboot' data to the main partition? To make it possible to boot OSx as usual?

Niresh12495, 2020-12-03:
@GoodBoris Bad idea because
Preboot is how apple boots its OS. So It is the native way.

edwardbelotserkovsky, 2020-12-03:
Well, previously I always have been choosing "Boot MacOS from OSx partition". I don't know what's was happening under the hood, but I've never seen the "Preboot" option before. I sent some time to extract that option by pressing F3. My main concern is to return the booting flow to the usual way.
Even more, OC doesn't display "Preboot" option in any case and can boot OSx from the "main" partition.

GoodBoris, 2020-12-03:
> @GoodBoris Bad idea because
> Preboot is how apple boots its OS. So It is the native way.

Well, previously I always have been choosing "Boot MacOS from OSx partition". I don't know what's was happening under the hood, but I've never seen the "Preboot" option before. I sent some time to extract that option by pressing F3. My main concern is to return the booting flow to the usual way.
Even more, OC doesn't display "Preboot" option in any case and can boot OSx from the "main" partition.

Niresh12495, 2020-12-04:
Yes open core only shows the main volume of the APFS container, a similar method would be highly appreciated

SergeySlice, 2020-12-09:
> Is there any special option to make boot from Preboot as default?
> I have “Mac” as Default partition to boot from, How to use Preboot of Mac as Default?

Sorry, but it works.
I set a system to start in System Preferences and push "Restart". It works and it works next reboot.
See your mistakes.

Erictoby, 2021-01-12:
What we want is a method to hide the "Mac" system volume and display only "Preboot" when preboot volume has been installed correctly since Big Sur use Preboot as a default volume to boot now. Can we hide "Mac" system volume as we could hide the preboot volume in config.plist/GUI/Hide? PS: I also had to remove config.plist/Boot/DefaultVolume as it points not the preboot but the "Mac" system volume.

LeeBinder, 2021-01-24:
> Sorry, but it works.
> I set a system to start in System Preferences and push "Restart". It works and it works next reboot.

@SergeySlice Does this also work for you booting Big Sur? Have you tested and verified this on one of your hack-Macs?

SergeySlice, 2021-02-06:
Yes, it is not possible to reboot to BigSur from Mojave because of different apfs systems.
Anyway what do you want? The issue is wrong. Preboot works as default selection.

LeeBinder, 2021-02-06:
@SergeySlice yes, I had learned that in the meantime, thanks for clarifying it.

@Niresh12495 simply write `Preboot` into Clover config.plist > Boot > Default Boot Volume, and Big Sur should boot automatically. At least that works for me (even though I personally have `Mojave` in there)

closed 2021-02-27 by SergeySlice