CloverHackyColor/CloverBootloader #316: Legacy custom entries rename broken since r5109 (switch to new theming engine)

LeeBinder, 2020-11-30:
Until r5108 (last version with old theming engine), Clover boot GUI would correctly display MBR  volume labels set in the GUI custom legacy section:

[image]

In this case, since Windows 10 boot volume is not the Windows install but the System-reserved volume, I first hid the actual Windows 10 volume in Clover's Hide Volume section, then relabeled the legacy System-reserved volume to Windows 10 via the string extracted from the boot log:

<details>
  <summary>click to expand</summary> 

  ```<key>Legacy</key>
			<array>
				<dict>
					<key>Ignore</key>
					<false/>
					<key>Hotkey</key>
					<string>W</string>
					<key>Hidden</key>
					<false/>
					<key>Volume</key>
					<string>HD(1,MBR,0x8F50DEE8,0x800,0xFA000)</string>
					<key>Disabled</key>
					<false/>
					<key>Type</key>
					<string>Windows</string>
					<key>FullTitle</key>
					<string>Windows 10</string>
				</dict>
			</array>
  ```
</details>

But r5109 broke Windows legacy boot completely  - the Windows bootloader was not recognized as Windows but as Mac:

[image]

and after selecting the Windows entry, Clover would crash.

This was only partially fixed with the r5115 update. Ever since the ability to boot into legacy Windows was restored, but the custom label remains broken, even with the most current r5127:

[image]

You see the gap between _Boot_ and _from_? That's where Clover should find the variable set as Title in config.plist, then display ONLY that title. But the variable content gets lost somewhere since r5115, resulting in what you see.

None of the following makes a difference:

- `Title` or `FullTitle`
- volume's label `System-reserviert` instead the boot string
- volume's UUID instead the boot string

On first glance this does not seem like a deep-down-the-rabbit-hole bug, hopefully :) ?

Let me know if you need any logs or so from me, and I'll gladly supply.

LeeBinder, 2020-12-13:
I see you still support legacy boot because for [v5.1 r5125](https://github.com/CloverHackyColor/CloverBootloader/releases/tag/5125), I see

`Resolved legacy boot (6,7) file.`

LeeBinder, 2020-12-25:
@chris1111 I'm shocked. You simply close my bug report without any comment of consideration. Why then carefully craft an issue, even narrowed down as much as possible? A rather uncool Xmas present :(

@jief666 or @SergeySlice or @Sher1ocks I kindly request this to be reopened: better be fixed next than ignored this year. It worked before so anything else but impossible? Thank you all.

chris1111, 2020-12-25:
Its just an error sorry don't shocked and uncool Xmas for this 😁 🌲

LeeBinder, 2020-12-26:
Mighty fine. Imagine it's just a misplaced comma or so from r5114 to r5115. I know it's just more or less cosmetic but sure not a biggy to restore, once tackled. Me patient & not in a hurry 🥇

jief666, 2021-04-20:
Could you send me this config.plist ?

LeeBinder, 2021-04-21:
yes sure jief - can do once I'm back home tomorrow.

LeeBinder, 2021-04-21:
here you go:

- [config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6354213/config.plist.zip)
- dmesg boot.log from Clover r5108 (displaying Windows Volume Title as supposed): [bootlog Clover r5108.txt.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6354215/bootlog.Clover.r5108.txt.zip)

Thanks for taking this on and fixing this regression. Let me know if you need any more logs from later Clover.

jief666, 2021-04-22:
Please try this : 
[CloverX64-2021-04-22-16-42-25-a1a27d2-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6358748/CloverX64-2021-04-22-16-42-25-a1a27d2-dirty-jief.zip)

LeeBinder, 2021-04-22:
Compliments, jief, you did it again - confirmed WORKING: label title displays correctly, and Windows 10 also boots just fine - thank you so much 👍 🥇 !

Now I need to figure out how to boot OC based Clover 5133 on this Laptop, with OpenRuntime.efi + Quirks etc. Right now I still get a KP, not yet having had time to look into it.

In case I fail: can you point me to the lines of code you changed so I can potentially integrate them into the Clover r5122 code and compile that one myself?

closed 2021-04-22 by LeeBinder

jief666, 2021-04-22:
Legacy.cpp:325.
Replace the if by       
```
if ((Volume->BootType != BOOTING_BY_PBR) &&
          (Volume->BootType != BOOTING_BY_MBR) &&
          (Volume->BootType != BOOTING_BY_CD)) {
```
See also my post https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?do=findComment&comment=2756200

Do you ever had a working OC ? If yes, I have a process where I send you "hybrid" clover using OC config.plist that'll tell us what settings you are missing.

LeeBinder, 2021-04-22:
Gotcha. Replace
```
if (/*(Volume->BootType != BOOTING_BY_PBR) && */
          (Volume->BootType >= BOOTING_BY_MBR) /*&&
          (Volume->BootType != BOOTING_BY_CD)*/ ) {
```
with your lines of code from above. 

> Do you ever had a working OC ?

yes, on my primary hackintosh Notebook I have Clover on my SSD's ESP as primary bootloader and OC on a separate FAT16 volume (labeled OpenCore) as secondary bootloader, both working fine (but I prefer Clover for several reasons I don't need to mention here).

On this secondary hackintosh Laptop I only have Clover 5108/ 5122 running - never tried OC.

> If yes, I have a process where I send you "hybrid" clover using OC config.plist that'll tell us what settings you are missing.

If that's a generic OC config.plist, yes please, I'm interested. I sure would like to be able to use the latest Clover even on my secondary hack.

LeeBinder, 2021-04-23:
> Do you ever had a working OC ? If yes, I have a process where I send you "hybrid" clover using OC config.plist that'll tell us what settings you are missing.

@jief666 great news: successfully booting OC 0.6.8 32bit now on my legacy MBR hackbook via:

**boot** (installed via `OpenCore-0.6.8-RELEASE/Utilities/LegacyBoot/BootInstall_IA32.tool`
EFI/BOOT/**BOOTIA32.efi**
EFI/OC/[**Config.plist**](https://github.com/CloverHackyColor/CloverBootloader/files/6363962/Config.plist.zip) (taken from someone with a Laptop whose basic hardware - chipset, CPU, GPU - are the same as mine, then simplified)

There is about a 10 second hang during boot so it still needs some refinement but hey, latest OC arch32 is booting 👍 . Going to try the 64bit version tonite.

So if you could send me the "hybrid" clover which uses an OC config.plist, I'll gladly receive it :)

jief666, 2021-04-23:
I don't think we have a Clover 32 bits. Although Clover should compiled fine in 32 bits, but I never did it.
If, on one particular computer, you have an OC 64bits working but can't make Clover working, I can help. But there is a lot to setup se we have to share an EFI via GitHub. It's too difficult to send you file one by one. After a while, a mistake happen and I think you test something and you test something else without knowing it.
To share an EFI : have a look here : https://jief-machak.gitbook.io/cloverhelp/
But first, once OC is working, you have to try to build a config.plist for Clover that mimics what's in you config.plist for OC : same ACPI patches, kext patches, same drivers, same kexts, etc, etc.

LeeBinder, 2021-04-24:
thank you @jief666 but I was able to narrow it down: it's a limitation of OpenRuntime.efi!

In OpenCore, there is two flavors of it, 64bit (the one also Clover uses) and 32bit. When I install OC 64bit, it's boot picker panics with !!!! x64 exception type !!!! which is exactly the same panic I get with any OC based Clover. It shouldn't because the T7600 Core 2 Duo IS 64bit beyond the slightest shadow of the doubt; but this Laptop's CHIPSET indeed is 32bit which is why it does not make sense to install a x64 OS, but rather run it with (Clover) arch=i386 / OC: KernelArch=i386 which allows the OS to boot in memory saving 32bit mode, but still allow execution of 64bit apps on-demand. Some nice architecture magic Apple concocted back then with OS X Lion.

When I overwrite the 64bit OpenRuntime.efi with the 32bit flavor -> no panic but no boot menu, either, neither with Clover nor with OC. I've also tried the latest OC 0.6.9 nightly from dortania -> same thing.

This old Laptop is not worth putting more time & energy into this. I'm totally fine with that I could contribute to finding and fixing this regression in Clover, and that I can boot Lion via either Clover up to r5122 or even most current OC.

Again, thanks for your assistance - aim reached and deserved & ready to move on to greater waters 👍

jief666, 2021-04-24:
> most current OC
You mean most current OC 32 bits ?

So yes, it might be because of OC that there is a crash. I'm interested to find at least where in the code the crash happen. But that'll some some time, and if the laptop is old and it's the only one to do that, I understand. New Clover won't bring anything new to the laptop !

LeeBinder, 2021-04-24:
yes, OC 32 bits - I added that in my post, + the links to the T7600 C2D @ ark.intel.com and cpu-world.com.