CloverHackyColor/CloverBootloader #317: r5127 stuck at [EB LOG:EXITBS:START] (still)

adaml75, 2020-12-02:
I have the same issue since 5122 (this is the last one that works for me). Quirks are configured. After Clover update, booting hangs at
[EB LOG:EXITBS:START]
++++++++++++++++++++++++
No configuration changes seem to fix the problem. I would love to try Big Sur, and configuring Open Core is a PITA. My Clover installation works perfectly well for me.

Niresh12495, 2020-12-04:
Probably bad Quirks configuration, try tweaking them.

peterkkk366, 2020-12-07:
Disable CFG Lock in Bios, if no this option, check Quirks to be set to Yes: AppleCpuPmCfgLock and AppleXcpmCfgLock.

darrenkhouston, 2020-12-07:
Maybe try this solution, it worked for me.

https://github.com/CloverHackyColor/CloverBootloader/issues/309#issuecomment-740180694

closed 2020-12-20 by SergeySlice