CloverHackyColor/CloverBootloader #321: Uneble to compile latest master

serdeliuk, 2020-12-09:
Hello,
Trying to enable USE_OC_SECTION_Misc i found that i am unable to build the clover.

I am receiving this error without any modifications 
error: type of 'OcStorageInitFromFs' does not match original declaration

The full error is,
make: [/home/marc/.0.0.hackint0sh/0.UEFI-devel/Clover5127/Build/Clover/RELEASE_GCC53/FV/Ffs/A17F4A89-5F19-444f-B7BE-48195E0575DBSmbiosGenDxe/SmbiosGenDxeOffset.raw] Error 1 (ignored)
/home/marc/.0.0.hackint0sh/0.UEFI-devel/Clover5127/OpenCorePkg/Include/Acidanthera/Library/OcStorageLib.h:104:1: error: type of 'OcStorageInitFromFs' does not match original declaration [-Werror=lto-type-mismatch]
 OcStorageInitFromFs (
 ^
/home/marc/.0.0.hackint0sh/0.UEFI-devel/Clover5127/OpenCorePkg/Library/OcStorageLib/OcStorageLib.c:215:1: note: 'OcStorageInitFromFs' was previously declared here
GNUmakefile:24224: recipe for target '/home/marc/.0.0.hackint0sh/0.UEFI-devel/Clover5127/Build/Clover/RELEASE_GCC53/X64/rEFIt_UEFI/refit/DEBUG/CLOVERX64.dll' failed
 OcStorageInitFromFs (
 ^
/home/marc/.0.0.hackint0sh/0.UEFI-devel/Clover5127/OpenCorePkg/Library/OcStorageLib/OcStorageLib.c:215:1: note: code may be misoptimized unless -fno-strict-aliasing is used
lto1: all warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/gcc returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status

Best regards,

Marc

serdeliuk, 2020-12-10:
I have tested 5123 and build ok, starting with 5124 all public sources have the same issue, could be something on my side?
i have performed following actions to build
git clone --recurse-submodules  https://github.com/CloverHackyColor/CloverBootloader . -b 5124
. ./edksetup.sh
./ebuild.sh

and the result is that error for all Clover versions greater than 5123 which is the only one which successfully build....

stevezhengshiqi, 2020-12-10:
@serdeliuk You can download the latest compiled version from the [GitHub Action page](https://github.com/CloverHackyColor/CloverBootloader/actions/runs/410166186). For the errors you met, I guess it's an OpenCore side compiling issue, and what you need to do is to delete and re-clone the whole Clover repository.

serdeliuk, 2020-12-11:
Hi @stevezhengshiqi , thanks, i tested to remove whole folder and re-clone, didn't helped. Also i have tested all Clover versions between 5124 to 5127 all of them have the same issue, i wonder how nobody observed that the public sources does not compile until now .... :)
I tested the precompiled beta version too and does not help as does not have the USE_OC_SECTION_Misc enabled and the SecureBootModel is not disabled, and i think the  misc section is not yet implemented in Clover's config.plist
I am able to install Big Sur 11.0.1 but not able to boot due the apple secure boot model..... boot after install ending with a kernel panic "Rooting from the live  fs of a sealed volume is not allowed on RELEASE version"

stevezhengshiqi, 2020-12-11:
> i wonder how nobody observed that the public sources does not compile until now .... :)

I beleive it’s a user side issue. I built Clover r5127 several times using buildme without any problem. Also, as GitHub Actions can build the project successfully, there should be no compile issue in this repository at all. You can check https://github.com/CloverHackyColor/CloverBootloader/runs/1522718108?check_suite_focus=true to see the compile detail.

ramseysd, 2020-12-11:
Hi

ramseysd, 2020-12-11:
Cc

serdeliuk, 2020-12-11:
@stevezhengshiqi could be an OS issue? What OS do you use? I wonder what kind of issue is, i am able to successfully build 5123 but no other newer version. I am able to build my efi apps with gnu-efi, as well as with edk2, i am able to build OpenCore's apps as CleanNvram i have a modified one which accept command line parameters, everything works as expected. 
Clover's wiki say Ubuntu 19 but i have 18.04, i don't think this is the issue, i will try 19 too just to be sure.... :)

serdeliuk, 2020-12-11:
I have successfully compiled Clover on a fresh Ubuntu 20 as i was unable to install Ubuntu 19 because they have some issues with their repositories and the install process fail, however now i am facing a black screen boot if i use my compiled version, did you had this issue previously? Is there anything i can do to fix this? Basically i cannot see anything on the screen and only power cycle works, no ctrl+alt+del

serdeliuk, 2020-12-19:
OK, the main issue seems to be the development environment, i was able to successfully build and run master branch on a Mojave with xcode 11 + command line tools.
Will be nice to update the wiki > building page to more details about how to build Clover

SergeySlice, 2020-12-20:
As far as I understand the issue is about compilation in Ubuntu. Correct issue title please.

serdeliuk, 2020-12-20:
@SergeySlice thanks, i changed a bit the title, actually i was unable to successfully create a working binary on few different  linux versions, tested on different CentOS versions as well as on different Ubuntu versions, the only successful build was under osx....

SergeySlice, 2021-01-03:
https://stackoverflow.com/questions/45047888/suppress-wlto-type-mismatch-warning-for-struct-with-flexible-array

 the bug has been fixed as of gcc 7.3.1 or earlier.

closed 2021-01-03 by SergeySlice