CloverHackyColor/CloverBootloader #322: Can't update 11.0.1 to 11.1

HandaTakuya, 2020-12-14:
[image]

[image]

Clover 5127

kwangle912, 2020-12-15:
Same as me
[image]

DecadentIpsum, 2020-12-16:
Same here. Is it possible to download this update manually. I checked the [downloads section](https://support.apple.com/downloads/macos) on Apple's site but it seems that it's available.

ryulove3, 2020-12-16:
Can't download 'installer app' with clover.
I used the 20C69 source on another Mac.
The update itself went well.
However, I found out that the disk structure of Big Sur installed using Clover is different from that of Open Core. You cannot download the app from the App Store, nor can it be updated with software update.
Is it because of this?

HandaTakuya, 2020-12-16:
> Can't download 'installer app' with clover.
> I used the 20C69 source on another Mac.
> The update itself went well.
> However, I found out that the disk structure of Big Sur installed using Clover is different from that of Open Core.
> There is no SSV(signed system volume).
> You cannot download the app from the App Store, nor can it be updated with software update.
> Is it because of this?

I can download and update app in App store but can't update macOS in setting

I try download macOS 11.1 from App store and install update 11.1 with macOS installer.

albertoacj, 2020-12-18:
I can download and start the update, but there is a error in process something in apfs  timeout

serdeliuk, 2020-12-19:
I was able to install big sur 11.0.1 and update to 11.1, i don't think it is related to Clover bootloader...... However, using Clover only though preboot partition (install and later boot) works....

HandaTakuya, 2020-12-20:
> I was able to clean install big sur 11.0.1 and update to 11.1, i don't think it is related to Clover bootloader...... However, using Clover only though preboot partition (install and later boot) works....
> At this stage...

I boot Big sur with preboot but can't download macOS update in setting

SergeySlice, 2020-12-20:
No problem with download and update.

closed 2020-12-20 by SergeySlice