CloverHackyColor/CloverBootloader #323: Kernel Panic on Rooting from the live fs

itsbohara, 2020-12-19:
Full error:

Rooting from the live fs of a sealed volume is not allowed on a RELEASE build
This is due to issues around Secure Boot boot being enabled in big sur.

As in opencore we have securebootmodel settings but when using clover stuck on error. 

How can i boot using Clover ?

serdeliuk, 2020-12-19:
Try to install and boot from preboot partition, only this way works at this moment

1Revenger1, 2020-12-21:
You could try updating OC, Clover's OC is horribly out of date by this point and they did fix the secure boot issue.

LeeBinder, 2021-01-23:
Any update on this? Still encountering this same issue with r5128. OC boots into Big Sur just fine with SecureBootModel set to j213 for my Laptop, matching SMBIOS.

Are you devs planning on incorporating OC's Security section at all in the short, medium or long run?

serdeliuk, 2021-01-24:
Hi @LeeBinder, Clover has SecureBootModel `Disabled` by default.
The kernel panic is not a Clover issue but an Apple boot model, stating with BigSur Apple decided to enforce preboot partition usage, so, only from preboot partition the boot process it is allowed, otherwise the "rooting from live... " panic is thrown...
The only way to mask this will be to hide the preboot partitions and mask the boot process, but this is not something nice.
Clover is nicer than any other bootloaders because do not mask any boot entries.

LeeBinder, 2021-01-24:
@serdeliuk  Gotcha!

Partial success: Big Sur 11.2 booting now via Clover r5128 -> pre-boot Big Sur GUI boot entry! However it takes several minutes until login screen - something is slowing boot down dramatically. But once logged in, working in Big Sur seems 100% normal as in Mojave/ Catalina. So something still needs refinement. In bootlog, above all endless repeats of

`kernel: (apfs) apfs_vfsop_unmount:2785: all done.  going home`: [BootLog.txt.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5863374/BootLog.txt.zip)

serdeliuk, 2021-01-24:
> @serdeliuk Gotcha!
>
> Partial success: Big Sur 11.2 booting now via Clover r5128 -> pre-boot Big Sur GUI boot entry! However it takes several minutes until login screen - something is slowing boot [...]

Yeeee, nice, enjoy
You should check your config and disable all 3 debug sections, especially the boot one, i think this is why the boot it is very slow.
If the debug it is already disabled you may want to enable verbose boot and see where is spending time 
That message is not an error, i think is some genius programmer at apple :) they have a lot of nice outputs....

LeeBinder, 2021-01-25:
I found "Debug" tick boxes in `Boot`, `ACPI` and `Kernel and Kext Pactches`, and all are disabled. config.plist attached so you can see if all looks as you like it or if you can find anything else: [config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5866683/config.plist.zip)

Additionally, error reporting files attached for completion:
• [EssentialsList-2021-01-24 22.47.15.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5866684/EssentialsList-2021-01-24.22.47.15.zip)
• [RunMe output - MacBook-Pro.fritz.box.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5866685/RunMe.output.-.MacBook-Pro.fritz.box.zip)

Boot is spending time at maaaanyyyy places.....

.. and serdeliuk, can you attach your config.plist so I can have a look/ compare? That would be great 👍

serdeliuk, 2021-01-25:
According to your config.plist it seems that you use an older version of Clover, or at least for config, or both clover and config, please update your Clover to latest version, and note, it is not enough to upgrade just Clover you should consider to upgrade the config as well.

Config files are not the same for everyone for sure mine will not work on your system, you should customize your own according to your hardware, drivers, etc... however, you should be able to find some configs on my github repo, the dell repo have a more recent Clover

LeeBinder, 2021-01-26:
SUCCESS :) !!
- updated from r5128 latest stable (the one in my attachments) to 5129 RC1 -> just one hang left: `kernel: IOKit Daemon (kernelmanagerd) stall[0], (60s): 'PXSX'`
- Googled, found and applied https://github.com/acidanthera/AirportBrcmFixup#please-pay-attention 💯 🥇 

Thank you Marc for your hints & support 👍

LeeBinder, 2021-01-27:
Marc, did you manage to create a GUI custom entry for Big Sur Preboot labeled "Big Sur" or alike so it does not always appear as "Mac OS X from Big Sur via Preboot" in Clover boot screen?

serdeliuk, 2021-01-30:
> Marc, did you manage to create a GUI custom entry for Big Sur Preboot labeled "Big Sur" or alike so it does not always appear as "Mac OS X from Big Sur via Preboot" in Clover boot screen?

Didn't tested as to me is not a big deal, but probably should be possible

SergeySlice, 2021-03-21:
Big Sur must be booted by preboot no matter of Secure Model.

closed 2021-03-21 by SergeySlice