CloverHackyColor/CloverBootloader #328: Fatal Error: Incompatible Version returned from ApfsDriverLoader.efi

GingkathFox, 2020-12-27:
I keep getting this error when starting Clover from rEFInd: `Fatal Error: Incompatible Version returned from ApfsDriverLoader.efi`

I got Clover to start after installing on my sd card, but when i attempted to start macOS Mojave it failed to start. It just displayed the apple logo with no progress bar under it.

preboot.log is below:

<details>
<pre>
0:100  0:100  MemLog inited, TSC freq: 1800001685
0:100  0:000  CPU was calibrated with ACPI PM Timer
0:100  0:000  CloverX64 : Image base = 0x74303000
0:100  0:000  
0:100  0:000  Now is 27.12.2020,  04:27:50 (GMT)
0:100  0:000  Starting Clover revision: 5127 (master, commit c9cfc183d) on Apple EFI
0:100  0:000  Build id: 20201124161448-c9cfc18-5127-dirty
0:100  0:000  Build with: [Args: -D NO_GRUB_DRIVERS_EMBEDDED -D LESS_DEBUG -t GCC53 | -D NO_GRUB_DRIVERS_EMBEDDED -D LESS_DEBUG --conf=/Users/sergey/src/CloverBootloader/Conf -D USE_LOW_EBDA -a X64 -b RELEASE -t GCC53 -n 5 | OS: 10.14.6]
0:100  0:000  SimpleTextEx Status=Unsupported
0:100  0:000  === [ GetDefaultSettings ] ======================
0:100  0:000  === [ Get Smbios ] ==============================
0:103  0:003  Type 16 Index = 0
0:103  0:000  Total Memory Slots Count = 2
0:103  0:000  Type 17 Index = 0
0:103  0:000  SmbiosTable.Type17->Speed = 1600MHz
0:103  0:000  SmbiosTable.Type17->Size = 4096MB
0:103  0:000  SmbiosTable.Type17->Bank/Device = BANK 0 DIMM0
0:103  0:000  SmbiosTable.Type17->Vendor = 0x80AD
0:103  0:000  SmbiosTable.Type17->SerialNumber = 0x00000000
0:103  0:000  SmbiosTable.Type17->PartNumber = 0x483943434E4E4E424A544D4C41522D4E5544
0:103  0:000  Type 17 Index = 1
0:103  0:000  SmbiosTable.Type17->Speed = 1600MHz
0:103  0:000  SmbiosTable.Type17->Size = 4096MB
0:103  0:000  SmbiosTable.Type17->Bank/Device = BANK 1 DIMM0
0:103  0:000  SmbiosTable.Type17->Vendor = 0x80AD
0:103  0:000  SmbiosTable.Type17->SerialNumber = 0x00000000
0:103  0:000  SmbiosTable.Type17->PartNumber = 0x483943434E4E4E424A544D4C41522D4E5544
0:103  0:000  Boot status=0
0:103  0:000  Running on: 'MacBookAir7,2' with board 'Mac-937CB26E2E02BB01'
0:103  0:000  === [ GetCPUProperties ] ========================
0:103  0:000  CPU Vendor = 756E6547 Model=306D4
0:103  0:000   The CPU supported SSE4.1
0:103  0:000  BrandString = Intel(R) Core(TM) i5-5350U CPU @ 1.80GHz
0:103  0:000   The CPU supported turbo
0:103  0:000  MSR 0x35               20004
0:103  0:000  MSR 0xE2 before patch 00000408
0:103  0:000  MSR 0xCE              0005053B_F3011200
0:103  0:000  MSR 0x1B0             00000000
0:103  0:000  FSBFrequency = 150 MHz, DMI FSBFrequency = 25 MHz, Corrected FSBFrequency = 150 MHz
0:103  0:000  MaxDiv/MinDiv: 18.0/5
0:103  0:000  Turbo: 27/27/27/29
0:103  0:000  Features: 0x77FAFBFFBFEBFBFF
0:103  0:000  Threads: 4
0:103  0:000  Cores: 2
0:103  0:000  FSB: 150 MHz
0:103  0:000  CPU: 2700 MHz
0:103  0:000  TSC: 2700 MHz
0:103  0:000  PIS: 600 MHz
0:103  0:000  ExternalClock: 150 MHz
0:103  0:000  === [ GetDevices ] ==============================
0:103  0:000  GOP found at: 
0:103  0:000  PCI (00|00:00.00) : 8086 1604 class=060000
0:103  0:000  PCI (00|00:02.00) : 8086 1626 class=030000
0:103  0:000   - GFX: Model=Intel HD Graphics 6000 (Intel)
0:103  0:000  PCI (00|00:03.00) : 8086 160C class=040300
0:103  0:000  PCI (00|00:14.00) : 8086 9CB1 class=0C0330
0:103  0:000  PCI (00|00:15.00) : 8086 9CE0 class=080102
0:103  0:000  PCI (00|00:15.04) : 8086 9CE6 class=0C8000
0:103  0:000  PCI (00|00:16.00) : 8086 9CBA class=078000
0:103  0:000  PCI (00|00:16.01) : FFFF FFFF class=FFFFFF
0:103  0:000  PCI (00|00:1B.00) : 8086 9CA0 class=040300
0:103  0:000  PCI (00|00:1C.00) : 8086 9C90 class=060400
[... 537 lines cut ...]
79:081  0:000  140  |  00  00  00  00  00  00  00  00  00  00
79:081  0:000  150  |  00  00  00  00  00  00  00  00  00  00
79:081  0:000  160  |  00  00  00  00  00  00  00  00  00  00
79:081  0:000  170  |  00  00  00  00  00  00  44  43  4E  37
79:081  0:000  180  |  33  35  32  50  53  38  58  46  34  34
79:081  0:000  190  |  33  41  4A  0A  20  20  20  20  20  00
79:081  0:000  200  |  00  00  00  00  00  00  00  00  00  00
79:081  0:000  210  |  00  00  00  00  00  00  00  00  00  00
79:081  0:000  220  |  00  00  00  00  00  00  00  00  00  00
79:081  0:000  230  |  00  00  00  00  00  00  00  00  00  00
79:081  0:000  240  |  00  00  00  00  00  00  00  00  00  00
79:081  0:000  250  |  00  00  00  00  00  00
79:081  0:000  Intel GFX injection not set
79:081  0:000  CurrentMode: Width=1440 Height=900
79:113  0:031  Use origin smbios table type 1 guid.
79:115  0:002  SetStartupDiskVolume:
79:115  0:000    * Volume: 'Macintosh HD'
79:115  0:000    * LoaderPath: ''
79:115  0:000    * DevPath: Macintosh HD
79:115  0:000    * GUID = 5EDAD956-A19C-496B-8E31-C97148E8AD4C
79:126  0:010    * efi-boot-device: <array><dict><key>IOMatch</key><dict><key>IOProviderClass</key><string>IOMedia</string><key>IOPropertyMatch</key><dict><key>UUID</key><string>5EDAD956-A19C-496B-8E31-C97148E8AD4C</string></dict></dict></dict></array>
79:140  0:014  Custom boot is disabled
79:141  0:000  Closing log
</pre>
</details>

SergeySlice, 2021-01-02:
Take ApfsDriverLoader.efi from official release to have a compatible version

GingkathFox, 2021-01-02:
official release?

SergeySlice, 2021-01-02:
https://github.com/CloverHackyColor/CloverBootloader/releases

SergeySlice, 2021-01-02:
But I may propose you problem related to rEFInd which is not supported at all.

GingkathFox, 2021-01-02:
yeah, i installed the latest release, thats where the error is coming from. and it may be rEFInd? ill have to see if i can boot directly into clover, i dont think it showed up when rebooting while holding option

closed 2021-01-15 by SergeySlice