CloverHackyColor/CloverBootloader #330: Kernel Panic - panic cpu2 caller ...initproc exited --exit reason namespace 2subcode 0x4 description:none

ManasMalla, 2020-12-30:
I have created a Clover EFI with r5127, with help of the guide given for big sur and added the necessary kexts, drivers and made the efi to match settings from my old EFI. I am on Pentium G4600, i was able to boot the macOS Catalina installer on r5122, but running big sur installer gives kernel panic -
panic cpu2 caller ...initproc exited --exit reason namespace 2subcode 0x4 description:none.
I gave the FakeCPUID 0x0306A0 in the config.plist with the help of clover configurator, added the FakePCIID.kext. Any help at the earliest is highly appreciatable.
I have been able to boot it using OpenCore but I wanted Clover as it helps me with my development process.

SergeySlice, 2021-01-07:
May be debug.log will show what is happen?

SergeySlice, 2021-03-21:
abandoned

closed 2021-03-21 by SergeySlice

mikvas63, 2021-03-21:
См