CloverHackyColor/CloverBootloader #331: Does not compile anymore

ProfDrLuigi, 2021-01-04:
Compiled fine in the past under 10.15.x. But now I get these Errors and the resulting PKG is not fully equipped.

[CP] BOOTX64
cp: /private/tmp/nightlies/clover/CloverPackage/CloverV2/EFI/BOOT/BOOTX64.efi: No such file or directory
[CP] CLOVERX64
cp: /private/tmp/nightlies/clover/CloverPackage/CloverV2/EFI/CLOVER/CLOVERX64.efi: No such file or directory
[CP] drivers/BIOS
cp: /private/tmp/nightlies/clover/CloverPackage/CloverV2/EFI/CLOVER/drivers/BIOS: No such file or directory
[CP] drivers/UEFI
cp: /private/tmp/nightlies/clover/CloverPackage/CloverV2/EFI/CLOVER/drivers/UEFI: No such file or directory

SergeySlice, 2021-01-05:
I see no error on your screen and I have no such a problem.

ProfDrLuigi, 2021-01-05:
Ok the Topic was not exactly written. Corrected it. 

Compiling works. But in the Endproduct are files missing. For example: CLOVERX64.efi and BOOTX64.efi

[image]

SergeySlice, 2021-01-07:
I compiled 5128 just now without problems.

ProfDrLuigi, 2021-01-07:
That is very strange. I did a fresh git clone and startet makeall.sh directly. For example your ISO Image is 10MB and my only 5MB. But why? Therer are a lot of not found folders in my log. Here the complete log:

```
bash-3.2$ bash makeall.sh 
TOOLCHAIN_DIR: /Users/luigi/test/CloverBootloader/toolchain

Status: cctools-949.0.1.tar.gz not found.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1922k  100 1922k    0     0  1786k      0  0:00:01  0:00:01 --:--:-- 1786k
- Creating new RAM disk

Initialized /dev/rdisk2 as a 300 MB case-insensitive HFS Plus volume
-  cctools-949.0.1 extract...
-  cctools-949.0.1 make mtoc...
-  cctools-949.0.1 installing mtoc...
-  cctools-949.0.1 mtoc installed in /Users/luigi/test/CloverBootloader/toolchain

- Ejecting RAM disk
"disk2" ejected.
MTOC_PREFIX: /Users/luigi/test/CloverBootloader/toolchain/bin/

Status: nasm-2.15.05.tar.xz not found.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  972k  100  972k    0     0   238k      0  0:00:04  0:00:04 --:--:--  238k
- Creating new RAM disk

Initialized /dev/rdisk2 as a 300 MB case-insensitive HFS Plus volume
-  nasm-2.15.05 extract...
-  nasm-2.15.05 configure...
-  nasm-2.15.05 make...
-  nasm-2.15.05 installing...
-  nasm-2.15.05 installed in /Users/luigi/test/CloverBootloader/toolchain

- Ejecting RAM disk
"disk2" ejected.
NASM_PREFIX: /Users/luigi/test/CloverBootloader/toolchain/bin/
NASM_VER: 2.15.05
Initializing workspace
WORKSPACE: /Users/luigi/test/CloverBootloader
EDK_TOOLS_PATH: /Users/luigi/test/CloverBootloader/BaseTools
CONF_PATH: /Users/luigi/test/CloverBootloader/Conf
Copying $EDK_TOOLS_PATH/Conf/build_rule.template
     to /Users/luigi/test/CloverBootloader/Conf/build_rule.txt
Copying $EDK_TOOLS_PATH/Conf/tools_def.template
     to /Users/luigi/test/CloverBootloader/Conf/tools_def.txt
Copying $EDK_TOOLS_PATH/Conf/target.template
     to /Users/luigi/test/CloverBootloader/Conf/target.txt
Building tools as they are not found
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C Source/C
Attempting to detect HOST_ARCH from 'uname -m': x86_64
Detected HOST_ARCH of X64 using uname.
mkdir -p .
mkdir ./libs 
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C Common
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  BasePeCoff.c -o BasePeCoff.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  BinderFuncs.c -o BinderFuncs.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  CommonLib.c -o CommonLib.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  Crc32.c -o Crc32.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  Decompress.c -o Decompress.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  EfiCompress.c -o EfiCompress.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  EfiUtilityMsgs.c -o EfiUtilityMsgs.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  FirmwareVolumeBuffer.c -o FirmwareVolumeBuffer.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  FvLib.c -o FvLib.o
gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-deprecated-declarations -Wno-self-assign -Wno-unused-result -nostdlib -g -O2  MemoryFile.c -o MemoryFile.o
[... cut ...]
-rw-r--r--   1 luigi  staff  1153672 Jan  7 10:14 CloverISO-5128.tar.lzma
-rw-r--r--@  1 luigi  staff  5296652 Jan  7 10:13 Clover_r5128.pkg
-rw-r--r--   1 luigi  staff       62 Jan  7 10:14 Clover_r5128.pkg.md5
drwxr-xr-x   3 luigi  staff       96 Jan  7 10:13 Resources
drwxr-xr-x   6 luigi  staff      192 Jan  7 10:13 build
drwxr-xr-x   3 luigi  staff       96 Jan  7 10:14 i386
drwxr-xr-x   3 luigi  staff       96 Jan  7 10:14 package
drwxr-xr-x   6 luigi  staff      192 Jan  7 10:13 utils
done!
bash-3.2$ 
```

SergeySlice, 2021-01-08:
I understand. You didn't include submodule
git clone --recurse-submodules ....

ProfDrLuigi, 2021-01-08:
Thx, that did the trick.

closed 2021-01-08 by ProfDrLuigi