CloverHackyColor/CloverBootloader #332: New Version Not work with Pre Bootloader

dateno1, 2021-01-05:
I was used https://github.com/ValdikSS/Super-UEFIinSecureBoot-Disk to bypass Secureboot for clover

bootx64.efi (shim) → grubx64.efi (preloader) → grubx64_real.efi (grub2) → Clover (bootx64.efi)

It work well with old version (tested with clover 5122) but lastest version (5127) has problem

I can't find macOSXs on boot menu (they are disappeared from menu)

SergeySlice, 2021-01-05:
Anyway debug.log may show what is absent and what is present.

dateno1, 2021-01-05:
> 
> 
> Anyway debug.log may show what is absent and what is present.

How can i get log?

dateno1, 2021-01-06:
> 
> 
> Anyway debug.log may show what is absent and what is present.

How can i get 'debug.log'?

I tried 'GUI-DebugLog' but i can't found any file on misc

also i was pushed F2 but 'preboot.log' not saved

SergeySlice, 2021-01-07:
Clover will save debug.log if the folder "CLOVER" is located on FAT32 partition, EFI partition.

dateno1, 2021-01-07:
> 
> 
> Clover will save debug.log if the folder "CLOVER" is located on FAT32 partition, EFI partition.

I was installed Clover on EFI partition (FAT32) but log not saved

If i success to boot by direct boot (without another boot loader) i can found 'preboot.log' on 'misc' from 'push F2 key'

stevezhengshiqi, 2021-01-08:
@dateno1 You need to enable Debug in config.plist.

dateno1, 2021-01-08:
> 
> 
> @dateno1 You need to enable Debug in config.plist.

Is that 'GUI' - 'DebugLog' - 'True'?

but That not worked (no 'debug.log' on 'misc')

stevezhengshiqi, 2021-01-08:
@dateno1 I don't think that's the one you need. It should be Boot - Debug, see https://github.com/CloverHackyColor/CloverBootloader/blob/master/CloverPackage/CloverV2/EFI/CLOVER/config-sample.plist#L384

dateno1, 2021-01-08:
> 
> 
> @dateno1 I don't think that's the one you need. It should be Boot - Debug, see https://github.com/CloverHackyColor/CloverBootloader/blob/master/CloverPackage/CloverV2/EFI/CLOVER/config-sample.plist#L384

I want to Bypass Secureboot

Parameter changed?

I tried https://sourceforge.net/p/cloverefiboot/wiki/FAQ/#q-i-installed-clover-but-i-just-get-a-black-screen

I will Test New parameter with v5128 after few hours after

dateno1, 2021-01-08:
[image]
I was added 'Debug' option but can't found any file on 'misc' (Tested with v5128)

https://blog.dateno1.com/?p=2072
My EFI fullset on here (sorrry korean only) (please download 'E3-1230_GA-B75M-D3H_EFI_BigSur_210107')

I was install pre-loader on boot (rename Clover's bootx64.efi and install that)

Can i use clover with Secureboot without Pre-Loader?

SergeySlice, 2021-01-15:
No, always use Preboot.

closed 2021-01-15 by SergeySlice

dateno1, 2021-01-15:
> 
> 
> No, always use Preboot.

What do you mean?

I can't find any macOSx entry from menu (normal&preboot) with pre-bootloader (Even press F3)