CloverHackyColor/CloverBootloader #341: EFI загрузчик не читает конфу из \EFI\CLOVER

dialmak, 2021-01-13:
EFI загрузчик не читает конфу из \EFI\CLOVER 
Я переименовал \EFI\BOOT\BOOTX64.EFI в \EFI\BOOT\grubx64_real.efi для обхода SECURE BOOT 
Стандартная ситуация, я так делаю давно, это прекрасно работает
https://i.imgur.com/SFttIvq.png
подробности тут 
https://github.com/ValdikSS/Super-UEFIinSecureBoot-Disk
https://habr.com/ru/post/446072/
Вопрос. Как заставить CLOVERX64.EFI переименованный в \EFI\BOOT\grubx64_real.efi читать конфу из \EFI\CLOVER?

EFI bootloader does not read conf from \EFI\CLOVER
I renamed \EFI\BOOT\BOOTX64.EFI to \EFI \BOOT\grubx64_real.efi to bypass SECURE BOOT
Standard situation, I've been doing this for a long time, it works great
https://i.imgur.com/SFttIvq.png
details here
https://github.com/ValdikSS/Super-UEFIinSecureBoot-Disk
https://habr.com/ru/post/446072/
Question. How can I get CLOVERX64.EFI renamed to \EFI\BOOT\ grubx64_real.efi to read the conf from \EFI\CLOVER?

SergeySlice, 2021-01-15:
Все как работало так и работает.
Где debug.log?

Drovosek01, 2021-01-22:
Если у тебя на этом компе стоит и Windows, то можешь в UEFI выбрать загрузку Windows, а в самой Windows отредактировать BCD, чтобы он запускал Clover.
У меня в китайской материнке x79 нет возможности выбрать .efi файл, с которого запускаться и так я решил эту проблему, может и вам поможет:
https://www.applelife.ru/threads/manual-ustanovka-macos-i-windows-na-odin-xard-poshagovo.38295/page-159#post-918772

dialmak, 2021-01-22:
> Где debug.log?
https://www.upload.ee/files/12798499/preboot.zip.html

SergeySlice, 2021-01-23:
Раз ты хочешь такие жонглирования в grub и иже с ними, давай уж сам разберись, кто куда и зачем лезет.
Кловер не предназначен для работы совместно с Грубом.
Если будут конкретные предложения, создай пул-реквест.

closed 2021-01-23 by SergeySlice

dialmak, 2021-01-23:
> Раз ты хочешь такие жонглирования в grub и иже с ними, давай уж сам разберись, кто куда и зачем лезет.
Причем здесь grub?  Весь смысл обойти Secure Boot. 
Ну да ладно - будем юзать r5122