CloverHackyColor/CloverBootloader #342: It seem CPU Emulate not work with new version

dateno1, 2021-01-14:
Hardware : G3930 + GA-H110-D3A + GT 630
OS : Catalina 10.15.7

Old (without OC) Version work well but Lastest (5128) Not worked it

NullCPUPowerManagement.kext & DummyPowerManagement both tested but not worked anything

I was success to boot with i3-7100 but G3930 is not (Just Change CPU&Can't Boot)

It will be KP (seem officially not supported CPU error)

'FakeCPUID -> Cpuid1Data' is worked with Clover?

SergeySlice, 2021-01-15:
Why not?

dateno1, 2021-01-15:
> 
> 
> Why not?

I don't know but it seem not worked with new version clover

G3930 + Old Clover + NullCPUPowerManagement.kext : OK
i3 + Old Clover : OK
G3930 + New Clover + NullCPUPowerManagement.kext : KP
G3930 + New Clover + DummyPowerManagement Quirks : KP
i3 + New Clover : OK

All other settings&kexts&drivers is same

SergeySlice, 2021-01-23:
As far as I know FakeCPUID in Clover works in new versions.

dateno1, 2021-01-24:
> 
> 
> As far as I know FakeCPUID in Clover works in new versions.

How do you success?

SergeySlice, 2021-01-27:
Try again.

closed 2021-02-14 by SergeySlice