CloverHackyColor/CloverBootloader #344: This doesn't feel right upgrading from 5122

cyberfunk, 2021-01-16:
The only way I can get versions greater than 5122 to work with my Clover setup is to copy and paste the quirks from the 5123+ version sample config.plist and delete DataHubdxe.efi driver from my EFI/Clover drivers folder. I have ransacked the internet and can not get a definitive answer on what DataHubdxe.efi does but it is probably there for a reason. I also have no idea what that new quirks section does and I do not feel comfortable just blindly copying and pasting them into my plist. Can anybody give me peace of mind and explain what the quirks section is doing to my system (that was already working perfectly under 5122) and also can I safely delete DataHubdxe.efi and not give it another thought?

Erictoby, 2021-01-22:
> The only way I can get versions greater than 5122 to work with my Clover setup is to copy and paste the quirks from the 5123+ version sample config.plist and delete DataHubdxe.efi driver from my [...]

Right. As the "quirks" came from opencore, the "[opencore install guide](https://dortania.github.io/OpenCore-Install-Guide/)"  explains them very well.  (i.e., desktop / laptop / chipsets).

cyberfunk, 2021-01-22:
> Right. As the "quirks" came from opencore, the "[opencore install guide](https://dortania.github.io/OpenCore-Install-Guide/)" explains them very well. (i.e., desktop / laptop / chipsets).

I'm not using OpenCore. I'm running Clover only so in theory I shouldn't even have to amend my plist with those quirks or remove drivers or learn about OpenCore. I just want a smooth trouble free upgrade and hope someone with answers can help me out😁

jief666, 2021-01-22:
Some of these quirks were setup by aptiomemoryfix.efi and equivalent.
Clover now use OpenCore kernel patcher. I'm afraid you'll have to configure these quirks. There is some default value but I doubt it could work out of the box.
Sorry, no smooth trouble free upgrade there....

cyberfunk, 2021-01-23:
> Some of these quirks were setup by aptiomemoryfix.efi and equivalent.
> Clover now use OpenCore kernel patcher. I'm afraid you'll have to configure these quirks. There is some default value but I doubt it could work out of the box.
> Sorry, no smooth trouble free upgrade there....

Is there some advantage to upgrading? My setup is rock solid on last release of Catalina with Clover 5122. OpenCore was unstable on the same setup which is why I reverted back to Clover. I don't need to upgrade I was just wondering why I was having issues if I did want to move forward. Thanks for your input:-)

SergeySlice, 2021-01-23:
DataHubDxe is still obligatory driver.

cyberfunk, 2021-01-23:
Thanks for confirming! When I upgraded Clover and removed DataHubDxe everything appeared to still be working but I did not want to risk a crash down the road when performing a DJ set due to an unforeseen conflict with the missing DataHubDxe. I think I'll just stay on 5122 for the foreseeable future.

closed 2021-01-23 by cyberfunk