CloverHackyColor/CloverBootloader #345: displayport working until clover 5122. Kext patches not working: displayport to hdmi

andresrobanesky, 2021-01-18:
Clover versions incorporating opencore 5124 "This release integrated OpenCore and delegates kernel and kext injection and patching to OC which allows to boot BigSur."  

The kext patches don't  work 

these patches don't work

AppleIntelSKLGraphicsFramebuffer.   find. 01030303.  replace.   01030403.  '10.11.4-SKL-1912000-Port_0x5-DP2HDMI'
AppleIntelSKLGraphicsFramebuffer. find. 02040A00 00040000 87010000.  replace. 02040A00 00080000 87010000
'10.11.4-SKL-1912000-Port_0x6-DP2HDMI'

This is my clover efi folder ...

[CLOVER.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5827405/CLOVER.zip)

if i use clover 5122 it works, but if i use later versions it doesn't work.

just replacing the cloverx64.efi file to a version later than 5122, the displayport does not work, keeping the other files

System version: catalina, 10.15.7

intel 530
Intel copre i7 6 gen 4ghz
board z170x gaming 6

I want to update to mac big sur, I already installed it, but the displayport does not work for me, only the hdmi, I use a converter from dp 1.2 to hdmi 2.0. By doing tests, it was evident that the problem is present in the kext patch that previously worked correctly. Please help!!.

I require the mac big sur with Displayport to have hdmi 2.0

SergeySlice, 2021-01-23:
Yes, graphics kext can't be patched this way because to successfully boot BigSur we have to use other kext loading procedure.
The solution can be using Lilu+WhateverGreen

andresrobanesky, 2021-01-26:
> Yes, graphics kext can't be patched this way because to successfully boot BigSur we have to use other kext loading procedure.
> The solution can be using Lilu+WhateverGreen

Thanks for your answer, but also with mac Catalina the error appears, not only big sur. the problem is opencore not patching the video kext, clover if it patches the video kext fine, i use lillu and whatevergreen in clover and opencore. 

the version 5124 of clover "This release integrated OpenCore and delegates kernel and kext injection and patching to OC which allows to boot BigSur."

i need this patch.
 10.11.4+-SKL-1912000-4_displays
AppleIntelSKLGraphicsFramebuffer
Find. 01030303
Replace 01030403

I've been researching on the internet how to make the display port work with opencore for the intel hd 530 and I can't find anything.

SergeySlice, 2021-01-27:
For Catalina you may use Clover 5123.1.

closed 2021-02-06 by SergeySlice