CloverHackyColor/CloverBootloader #347: Вопрос. У меня вначале вначале в меню значки из Custom и за ними значки из Scan.

dialmak, 2021-01-23:
У меня вначале вначале в меню значки из Custom и за ними значки из Scan.
Как сделать наоборот?

SergeySlice, 2021-01-27:
Наверно никак.
Custom так пускай все будут кастом.

dialmak, 2021-01-28:
Жаль. У меня 15 Custom. В rEFInd сканированные добавляются в начале.
Еще одно. А можно ли загрузить пользовательский config_user.plist из config.plist?
К примеру я так гружу пользовательскую конфу efitools.conf в rEFInd из refind.conf (ключ -c), [тут гифка](https://137615.selcdn.ru/NuttyBoot/efitools.gif)
```
menuentry "Management UEFI Boot Loaders" {
	volume "NUTTYBOOT"
	icon   \EFI\BOOT\icons\efiboot.png
	loader \EFI\tools\shellx64.efi
	options "-nostartup -noversion -nomap -exit \EFI\BOOT\refindx64.efi -c \EFI\BOOT\efitools.conf"
}
```

SergeySlice, 2021-01-30:
Видимо нужно создавать Custom Entry для этого. Там тоже есть и иконки и названия. При чем тут config_user? В какой момент вы собираетесь загружать его? До входа в ГУИ? Если после, то уже поздно, ГУИ уже нарисовано.
Всегда? Тогда можно просто включить в конфиг.

dialmak, 2021-01-30:
Вы меня не поняли, в гифке вроде бы все понятно. 

> ГУИ уже нарисовано.

Да, нарисовано. Если б можно было запускать config_user.plist из основного config.plist, то я сократил бы число Custom в основном меню перенеся их в config_user.plist, как и показано в гифке.

SergeySlice, 2021-02-06:
Зачем плодить файлы? Оно что, в один файл не помещается?
Вот зачем мне вводить новые, глючные алгоритмы в Кловере по такому странному запросу?

closed 2021-02-21 by SergeySlice