CloverHackyColor/CloverBootloader #348: Kexts file cannot be loaded normally after renaming

liliuwei, 2021-01-27:
Kexts file cannot be loaded normally after renaming, Clover5122 version is normal, but the latest Clover5128 is abnormal, for example: AppleALC.v1.5.6.kext, SMCProcessor.v1.1.9.kext;

However, it is normal for some non-core Kexts files with version numbers or duplicate names, for example: VoodooHDA.v2.9.6.kext

SergeySlice, 2021-01-27:
This is because 5124+ uses strict names for Lilu and plugins else they don't work.
Other kext have no such a problem.

liliuwei, 2021-01-28:
> 这是因为5124+对Lilu和插件使用严格的名称，否则它们将不起作用。
> 其他的kext没有这样的问题。

Thank you for your answer

liliuwei, 2021-01-28:
> This is because 5124+ uses strict names for Lilu and plugins else they don't work.
> Other kext have no such a problem.

Thank you for your answer

closed 2021-01-30 by SergeySlice