CloverHackyColor/CloverBootloader #349: [BUG] GUI Custom Entry for Big Sur Preboot breaks loading kexts from '11' folder

LeeBinder, 2021-01-28:
**Tested Clover versions:** 5128, 5129

**Expected behavior:** regardless if Big Sur is started via the default "Boot Mac OS X from Big Sur via Preboot" or via a custom GUI entry, kexts should be loaded from both locations.
However, having moved kexts from 'Other' into '11' (for compatibility with other versions of macOS installed parallel to other disk volumes which load their kexts from their respective folders) -> Kernel panic.

**Successless attempts to resolve or work-around:**
- created folders that appeared relevant (11.2/ 11. / 10.16) and placed kexts into those folders
- altered OS X 'Type' entry from Big Sur GUI entry: removed, Other, Preboot

**Steps to reproduce:**
- follow @vaczi001's [manual to create a custom GUI entry for Big Sur](https://github.com/CloverHackyColor/CloverBootloader/issues/300#issuecomment-768300847)
- move all kexts from OTHER into 11
- try boot Big Sur entry

vaczi001 (github nic) aka kushwavez (insanelymac nic) has [reproduced this issue on his Hack](url), too, also with Clover 5129. In addition he found out that

> From log I can see that seems like kexts are being searched and added to the array, but won't load after that:
```
0:993  0:000  AddKextsInArray from Other
0:993  0:000  AddKextsInArray from Off
0:993  0:000  AddKextsInArray from Unknown
```

> There is no "AddKextsInArray from 11" so the kexts won't load. 

Let me know if you need any more data from me.

serdeliuk, 2021-01-30:
@LeeBinder you need to enable debug log and provide us with the log file after you boot to bigsur in order to see what is going on, worth to mention that the 11.x folder should be 11.1 for BigSur 11.1 and should be outside "other" folder for example 
EFI/CLOVER/kexts/11.1

vaczi001, 2021-01-30:
@serdeliuk Unfortunately the insanelymac link he posted is broken, so I copy & paste the whole reply here, also to not bounce from here to insanelymac unnecessarily.

**[Comment from Insanelymac (kushwavez)](https://www.insanelymac.com/forum/topic/346498-gui-custom-entry-for-big-sur-preboot-how-to/?tab=comments#comment-2749617)**
with Custom entry our kexts are won't load from CLOVER/kexts/11
From log I can see that seems like kexts are being searched and added to the array, but won't load after that.
```
0:826  0:000  === [ InitKextList ] ============================
0:831  0:005  Added Kext=11\VoodooRMI.kext
0:832  0:001  Added Kext=11\VoodooSMBus.kext
0:833  0:000  Added Kext=11\AirportBrcmFixup.kext
0:833  0:000  Added Kext=11\AppleBacklightSmoother.kext
0:834  0:000  Added Kext=11\BrightnessKeys.kext
0:834  0:000  Added Kext=11\HibernationFixup.kext
0:835  0:000  Added Kext=11\IntelMausi.kext
0:835  0:000  Added Kext=11\NVMeFix.kext
0:836  0:000  Added Kext=11\Sinetek-rtsx.kext
0:836  0:000  Added Kext=11\SMCBatteryManager.kext
0:836  0:000  Added Kext=11\SMCLightSensor.kext
0:837  0:000  Added Kext=11\SMCProcessor.kext
0:838  0:000  Added Kext=11\SMCSuperIO.kext
0:838  0:000  Added Kext=11\USBPorts.kext
0:839  0:000  Added Kext=11\VirtualSMC.kext
0:839  0:000  Added Kext=11\YogaSMC.kext
0:840  0:000  Added Kext=11\VoodooPS2Controller.kext
0:841  0:001  Added Kext=11\WhateverGreen.kex
```
```
0:993  0:000  AddKextsInArray from Other
0:993  0:000  AddKextsInArray from Off
0:993  0:000  AddKextsInArray from Unknown
```

There is no "**AddKextsInArray from 11**" so the kexts won't load.
 
I think the reason is because of the "config.plist/GUI/Custom/Big Sur/Type=OSX" -> I think this can't determine what is the OS version, while if I boot for example from Recovery it's working because no Type is specified, and the OS version is determined by Clover automatically.
[2021-1-28_8-21-27_CLOVERX64.efi.log](https://github.com/CloverHackyColor/CloverBootloader/files/5897609/2021-1-28_8-21-27_CLOVERX64.efi.log)

serdeliuk, 2021-01-30:
I have tested on my side and work well, with this section in config.plist

`                                <dict>
                                        <key>Disabled</key>
                                        <false/>
                                        <key>FullTitle</key>
                                        <string>BigSur11.1</string>
                                        <key>Hidden</key>
                                        <false/>
                                        <key>Ignore</key>
                                        <false/>
                                        <key>Path</key>
                                        <string>\4B02232C-76BD-425D-AA97-C9D2D164948E\System\Library\CoreServices\boot.efi</string>
                                        <key>Type</key>
                                        <string>OSX</string>
                                        <key>Volume</key>
                                        <string>FB7CDCFA-3B77-40DE-A461-422C390022FD</string>
                                        <key>InjectKexts</key>
                                        <true/>
                                        <key>NoCaches</key>
                                        <false/>
                                </dict>
`
UUIDS can be found in preboot.log generated with F2 searching for `'Preboot'` string

`1:342  0:000  - [10]: 'Preboot'
1:343  0:001  contentDetails name:BigBenSur
1:343  0:000      AddLoaderEntry for Volume Name=Preboot, idx=7
1:343  0:000      ApfsContainerUUID=32B30F49-7F79-469A-BC93-FE27F1564EF0
1:343  0:000      ApfsFileSystemUUID=**FB7CDCFA-3B77-40DE-A461-422C390022FD**
1:343  0:000      APFSTargetUUID=**4B02232C-76BD-425D-AA97-C9D2D164948E**
1:344  0:000      Check if volume Is Hibernated:
1:344  0:000      Check sleep image 'by signature':
1:344  0:000      using default sleep image name = \private\var\vm\sleepimage
1:344  0:000      sleepimage not found -> Not Found
1:344  0:000      hibernated: no - sign
1:344  0:000      IconName=bigsur,mac comma=6 size=10
1:344  0:000      first=os_bigsur
1:344  0:000      Second=os_mac
1:344  0:000      Loader entry created for 'PciRoot(0x0)\Pci(0x1F,0x2)\Sata(0x0,0x0,0x0)\HD(5,GPT,8CCD0BA8-34D7-46DC-B201-1C071E6B2987,0x63FF38
00,0xA123160)\VenMedia(BE74FCF7-0B7C-49F3-9147-01F4042E6842,FADC7CFB773BDE40A461422C390022FD)\4B02232C-76BD-425D-AA97-C9D2D164948E\System\Libra
ry\CoreServices\boot.efi'
`

vaczi001, 2021-01-30:
@serdeliuk yes I have no problem with that either.
The problem we reported is not with that. The problem is: kexts from kexts/11 won't load

serdeliuk, 2021-01-30:
Sorry, did not realised that is about kexts versioning. Yes, tested and seems that kexts does not load indeed, from 11.1 in my case, let me check on my side with debug enabled will be faster...

LeeBinder, 2021-01-30:
@serdeliuk of course, `CLOVER/kexts/11` respectively `11.2` for my RC install. I refined my initial posting accordingly so it should be clear now that I know what I'm doing .. ;)

I also attached my Clover debug log to the first posting (and added the link to insanelymac for completion).

serdeliuk, 2021-01-30:
So far it seems that if you add a custom entry clover is unable to find the os version and does not know if it is required to load or no different kexts than the ones from "Other"  because does not have an OS version don't search in other places.
As far as i can see there is no way at this point to add to the custom entry a new filed to set "by hand" the os version, i still digging, i will let you know if i find anything

jief666, 2021-01-30:
Try this : 
[CloverX64-2021-01-30-17-49-47-b5e0562-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5898148/CloverX64-2021-01-30-17-49-47-b5e0562-dirty-jief.zip), that should solve the custom entry version detection problem, that led to kext loaded from Unknown instead of 11 (or any other).

jief666, 2021-01-30:
Please, tell me here if that works.
Thanks.

serdeliuk, 2021-01-30:
@jief666 thanks, this version identify correctly the OSVersion but does not boot, on my side my notebook entered in shutdown state after a while, also everything seems to have a lot of delay, did you added some kind of extra delay in debug loops?
Attached preboot.log and debug.log
[2021-1-30_17-1-11_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/5898378/2021-1-30_17-1-11_BOOTX64.EFI.log)
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/5898379/preboot.log)

GetOSVersion: 11.1 (20C69)
Filtering KernelPatches:
 - [00]: Disable panic kext logging on 10.14 Release kernel :: [OS: 11.1 | MatchOS: 10.14 | MatchBuild: no] ==> disabled by user
 - [01]: Disable panic kext logging on 10.13 Release kernel :: [OS: 11.1 | MatchOS: 10.13 | MatchBuild: no] ==> disabled by user
 - [02]: Lapic panic patch example :: [OS: 11.1 | MatchOS: All | MatchBuild: no] ==> disabled by user
 - [03]: Symbolic patch example got lapic panic :: [OS: 11.1 | MatchOS: All | MatchBuild: no] ==> disabled by user

vaczi001, 2021-01-30:
> Try this :
> [CloverX64-2021-01-30-17-49-47-b5e0562-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5898148/CloverX64-2021-01-30-17-49-47-b5e0562-dirty-jief.zip), that [...]

It's working (at least for me)! All of my kexts are loaded correctly from kexts/11 with a Custom entry.

[2021-1-30_17-58-53_CLOVERX64.efi.log](https://github.com/CloverHackyColor/CloverBootloader/files/5898445/2021-1-30_17-58-53_CLOVERX64.efi.log)

LeeBinder, 2021-01-30:
@jief666 mighty fine, you rock - all fine here now: most kexts in Other, some in 10.13/14/15,11. I've booted all my macOS 10.13 - 11.2 in the past hour, worked/tinkered this and that in all of them -> 👍 

What I'm wondering about is that you took the older 5128 source code as base, not the recent 5129:

[image]

@serdeliuk even though in theory it should not be necessary, it still can help to run hackintool's kext cache rebuild & perms repair util. At least for me it still tends to resolve oddities like those you describe.

vaczi001, 2021-01-30:
It's just a test CLOVER.efi @jief666 made for us for testing purpose, so it doesn't matter, and the code is the same as on v5129. The corrected code will be added in the next release (I think). It's cosmetic.

LeeBinder, 2021-01-30:
@vaczi001 boy am I relieved

[image]

serdeliuk, 2021-01-31:
@LeeBinder, indeed, i have too a  CleanNvram.efi special crafted for Clover (reboot when end) with a custom boot entry, sometimes helps too.

@jief666 if i disable boot -> debug works perfectly.
But with boot->debug enabled take approx 2min to get to Clover menu and after i chose to boot then another 3min reach "EB|LOG:EXITBS:START" then shutdown. 
For me this is not an issue as i do not need custom crafted boot to hide preboot text.

jief666, 2021-01-31:
The version number is just cosmetic. The commit number shows that it's the last code.
I'm closing the issue as it seems resolved. We can still continue to talk about it, if needed be.

closed 2021-01-31 by jief666