CloverHackyColor/CloverBootloader #351: Animations and time running very fast

xenonstride, 2021-01-29:
Hello,
I am using i5-1035G1 (Ice Lake) and there seems to be some issue with fsb or bus speed due to which the animations and time are running very fast. I have already tried setting the clock speed and bus speed manually with clover from 'cpu' section. What can I do to fix it? This issue is only present on clover but not on opencore. But I prefer clover because it is giving me low pkg power.

SergeySlice, 2021-01-30:
Please provide Clover's preboot.log to see what is happen.

xenonstride, 2021-01-30:
@SergeySlice here is the preboot.log
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/5898267/preboot.log)

LeeBinder, 2021-01-30:
If you don't already, try https://github.com/acidanthera/CpuTscSync. It does more than just solve KPs after wake.

xenonstride, 2021-01-31:
@LeeBinder Already tried it but the problem is still there.

xenonstride, 2021-01-31:
Here is my EFI and some videos of the problem I am describing
[EFI.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5899229/EFI.zip)

1 - https://drive.google.com/file/d/1Py1G8nzTc6tQMe1nB3-XGMndpKT7sFGk/view?usp=drivesdk
2 - https://drive.google.com/file/d/1y5KZbiuCpyI-dxgYOg8W1UtPmFOCk6Gy/view?usp=drivesdk
3 - https://drive.google.com/file/d/13KeV6NcC9qqPACoDtnYpuQShx7cPzXHv/view?usp=drivesdk

SergeySlice, 2021-01-31:
Can you provide also AIDA64 CPU report from Windows?

xenonstride, 2021-01-31:
@SergeySlice Here it is
[aida64-report.txt](https://github.com/CloverHackyColor/CloverBootloader/files/5899273/aida64-report.txt)

xenonstride, 2021-01-31:
> @SergeySlice Here it is
> [aida64-report.txt](https://github.com/CloverHackyColor/CloverBootloader/files/5899273/aida64-report.txt)

and also it seems like when I use Clover the max frequency of my cpu is only 2.3ghz whereas the actual maximum frequency of my cpu is 3.6ghz. in OpenCore it reaches 3.6ghz.

SergeySlice, 2021-02-05:
Check commit e24ceaa.

xenonstride, 2021-02-06:
@SergeySlice can you also tell me how to build from the latest commit?

xenonstride, 2021-02-06:
@SergeySlice Just tried the build from https://github.com/CloverHackyColor/CloverBootloader/actions/runs/541370980. The problem is still present.

SergeySlice, 2021-02-06:
I see no error in your log but I am not sure if used kexts are right for this CPU
1:207  0:000  Added Kext=Other\CPUFriend.kext
1:209  0:002  Added Kext=Other\CPUFriendDataProvider.kext
1:209  0:000  Added Kext=Other\CpuTscSync.kext
1:213  0:000  Added Kext=Other\SMCProcessor.kext
1:216  0:001  Added Kext=Other\VoltageShift.kext
And I don't know what you set in CPU section.
My proposition
BusSpeedkHz=99200
Type=0x0609
UseARTFrequency= NO

SergeySlice, 2021-02-06:
I have new idea. For this processor you have to set
~~~~
            <key>UseARTFrequency</key>
            <true/>

~~~~

SergeySlice, 2021-02-06:
DataHubDxe.efi must be present

xenonstride, 2021-02-07:
> I have new idea. For this processor you have to set
> 
> ```
>             <key>UseARTFrequency</key>
>             <true/>
> ```

Thank you so much @SergeySlice . This fixed it. All good now.

closed 2021-02-07 by xenonstride