CloverHackyColor/CloverBootloader #353: Only minimum CPU frequency in r5129

viniciusmi00, 2021-01-31:
After I upgraded to the Clover r5129, my CPU is only at the minimum frequency. I don't know if this is a bug or it was a planned change on the Clover r5129, so I will leave the log and a print screen here, if they help you.
At r5128 or less, it works well. 
CPU: i7 3930k
MB: Asus Rampage IV Extreme
Big Sur 11.1

[bootlog.txt](https://github.com/CloverHackyColor/CloverBootloader/files/5899299/bootlog.txt)

[image]

SergeySlice, 2021-02-05:
Check commit e24ceaa. Should resolve the issue.

viniciusmi00, 2021-02-11:
> Check commit [e24ceaa](https://github.com/CloverHackyColor/CloverBootloader/commit/e24ceaa47036b1b22de765a667641c9f3ae21d10). Should resolve the issue.

Did not work
I compiled using the "buildme" (SHA e24ceaa47). I used the option "7) build all". I installed it through "Clover_r5129.pkg" that was generated in the CloverPackage -> sym folder.
Is correct? 

[image]

SergeySlice, 2021-02-14:
But if you do 
~~~
Note: I have an SSDT created by "ssdtPRGen" i
~~~

then it is not a Clover issue.
Anyway I have to propose to check latest commit.
As well I may recommend to set FakeCPUID=0x0306A2

viniciusmi00, 2021-02-15:
> But if you do
> 
> ```
> Note: I have an SSDT created by "ssdtPRGen" i
> ```
> 
> then it is not a Clover issue.
> Anyway I have to propose to check latest commit.
> As well I may recommend to set FakeCPUID=0x0306A2

Slice, Clover r5130 release solve the problem, without me having to put FakeCPUID.
Problem solved!

closed 2021-02-15 by viniciusmi00