CloverHackyColor/CloverBootloader #354: Usage as Efildr replacement

LittleFox94, 2021-02-02:
I'm trying to use Clover as Efildr/DUET replacement, only using the boot sector (boot0af, boot1f32) and boot6. I was led to clover from https://gitlab.com/tianocore_uefi_duet_builds/tianocore_uefi_duet_installer, stating to use it instead of the old DUET builds.

I would like to have it boot `EFI\BOOT\BOOTX64.EFI`. Currently it only loads `EFI\CLOVER\CLOVERX64.EFI`. I propose falling back to the EFI standard removable media file if clover does not exist on the ESP.

Placing my custom loader at `EFI\CLOVER\CLOVERX64.EFI` already works fine, so it's only the matter of the path.

SergeySlice, 2021-02-05:
It is not a bug. boot6 made to boot CLOVERX64.EFI to not mix with EFI\BOOT\BOOTX64.EFI.
Sorry!

closed 2021-02-05 by SergeySlice

LittleFox94, 2021-02-05:
Thought of this as a feature request ^^'

Would you accept a pull request that loads `EFI\BOOT\BOOTX64.EFI` if `EFI\CLOVER\CLOVERX64.EFI` does not exist?

LittleFox94, 2021-02-21:
@SergeySlice just to make sure you saw my last comment - a no is fine :)

josephsmendoza, 2021-07-15:
@LittleFox94 could you describe to me the process of getting those files installed? I am trying to do the same thing with no success.

LittleFox94, 2021-07-15:
Am a bit busy right now with trying to heal and it's some time ago now, but maybe the commit I made already helps?

https://praios.lf-net.org/littlefox/lf-os_amd64/-/commit/8ef7f9c8c9b42e978eb41ba2526b3b7a55b72ace

Especially `images.cmake` should have some useful things

Feel free to ask again if you need more info, I can try to write some more things (:

josephsmendoza, 2021-07-15:
I'm not sure what I'm looking at with that commit. Thanks for the quick reply! But I was asking about what to do with the boot0af, boot1f32 and boot6 files to get DUET working.

LittleFox94, 2021-07-15:
OK, some more then, hope that helps ^^'

[https://praios.lf-net.org/littlefox/lf-os_amd64/-/blob/8ef7f9c8c9b42e978eb41ba2526b3b7a55b72ace/cmake/images.cmake#L17](In this line), I make a new fat32 filesystem in a previuosly created, but empty file using `mformat` (from `mtools`, userspace `FAT*` utilities). I give `boot1f32` as bootsector code to `mformat`. You could also patch an existing `FAT32` boot sector to include `boot1f32` as boot code (can link details if you need that, but would suggest using `mformat`).

In line 20 of that file, I copy `boot6` inside the filesystem, placed at `/boot`. Line 21 seems to use `boot0af` as code in `MBR`, patching it in with dd, while being careful about partition table (or rather protective MBR in this case, since it's GPT).

Here the relevant commands without cmake stuff, maybe making them easier to read.

```
dd if=/dev/zero of=hd.img bs=512 count=${image_size_hd_sectors}
sgdisk  -Z -o -n 1:${bootfs_start_sectors}:${image_size_headers} -t 1:ef00 hd.img -A 1:set:2
mformat -i hd.img@@1M -F -L ${bootfs_sectors} -H ${bootfs_start_sectors} -B clover/boot1f32
mcopy   -i hd.img@@1M -sbnmv clover/boot6 ::/boot
dd if=clover/boot0af of=hd.img bs=1 count=440 conv=notrunc
```

The variables are for disk image size, partition size, partition offset, .. all that stuff - calculation for these is in the same file linked above, starting at line 7 - I think that's readable enough and more a side note than the actual needed things, but if I'm wrong I can try explaining it more.

LittleFox94, 2021-07-15:
It will only boot an efi binary from the clover install path, not default efi path - be aware of that ^^