CloverHackyColor/CloverBootloader #357: Need more detailed log for error

DanielYWoo, 2021-02-06:
Upgrade Clover by installing r5129, it reports an error without any details, then I looked into the install.log, all I can find is "BiosBoot.pkg" encounters some problems, but no details. Can we put more details into install.log when something is wrong?

```
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: preinstall: Path to installer....... /Users/daniel/Downloads/Clover_r5129.pkg
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: preinstall: Path to destination..... /
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: preinstall: Path to dest volume..... /
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: preinstall: Root of system folder... /
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: rm: /Volumes/iMac/usr/standalone/i386/x64/boot6: Read-only file system
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: rm: /Volumes/iMac/usr/standalone/i386/x64/boot7: Read-only file system
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: rm: /Volumes/iMac/usr/standalone/i386/boot0af: Read-only file system
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: rm: /Volumes/iMac/usr/standalone/i386/boot0ss: Read-only file system
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: rm: /Volumes/iMac/usr/standalone/i386/boot1f32: Read-only file system
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: rm: /Volumes/iMac/usr/standalone/i386/boot1f32alt: Read-only file system
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: rm: /Volumes/iMac/usr/standalone/i386/boot1h: Read-only file system
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: rm: /Volumes/iMac/usr/standalone/i386/boot1h2: Read-only file system
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: rm: /Volumes/iMac/usr/standalone/i386/boot1x: Read-only file system
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: rm: /Volumes/iMac/usr/standalone/i386/boot1xalt: Read-only file system
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: rm: /Volumes/iMac/usr/standalone/i386/Description.txt: Read-only file system
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: ./preinstall: rm: /Volumes/iMac/usr/standalone/i386/Installation.txt: Read-only file system
2021-02-06 11:23:47+08 Daniels-iMac package_script_service[22607]: Responsibility set back to self.
2021-02-06 11:23:47+08 Daniels-iMac install_monitor[22606]: Re-included: /Applications, /Library, /System, /bin, /private, /sbin, /usr
2021-02-06 11:23:48+08 Daniels-iMac installd[22590]: PackageKit: releasing backupd
2021-02-06 11:23:48+08 Daniels-iMac installd[22590]: PackageKit: allow user idle system sleep
2021-02-06 11:23:48+08 Daniels-iMac installd[22590]: PackageKit: Cleared responsibility for install from 22586.
2021-02-06 11:23:48+08 Daniels-iMac installd[22590]: PackageKit: Cleared permissions on Installer.app
2021-02-06 11:23:48+08 Daniels-iMac installd[22590]: PackageKit: Install Failed: Error Domain=PKInstallErrorDomain Code=112 "An error occurred while running scripts from the package “Clover_r5129.pkg”." UserInfo={NSFilePath=./preinstall, NSURL=file:///Users/daniel/Downloads/Clover_r5129.pkg#BiosBoot.pkg, PKInstallPackageIdentifier=org.clover.biosboot, NSLocalizedDescription=An error occurred while running scripts from the package “Clover_r5129.pkg”.} {
	    NSFilePath = "./preinstall";
	    NSLocalizedDescription = "An error occurred while running scripts from the package \U201cClover_r5129.pkg\U201d.";
	    NSURL = "file:///Users/daniel/Downloads/Clover_r5129.pkg#BiosBoot.pkg";
	    PKInstallPackageIdentifier = "org.clover.biosboot";
	}
2021-02-06 11:23:48+08 Daniels-iMac installd[22590]: PackageKit: Running idle tasks
2021-02-06 11:23:48+08 Daniels-iMac installd[22590]: PackageKit: Done with sandbox removals
2021-02-06 11:23:48+08 Daniels-iMac Installer[22586]: install:didFailWithError:Error Domain=PKInstallErrorDomain Code=112 "An error occurred while running scripts from the package “Clover_r5129.pkg”." UserInfo={NSFilePath=./preinstall, NSURL=file:///Users/daniel/Downloads/Clover_r5129.pkg#BiosBoot.pkg, PKInstallPackageIdentifier=org.clover.biosboot, NSLocalizedDescription=An error occurred while running scripts from the package “Clover_r5129.pkg”.}
2021-02-06 11:23:48+08 Daniels-iMac installd[22590]: PackageKit: Removing client PKInstallDaemonClient pid=22586, uid=501 (/System/Library/CoreServices/Installer.app/Contents/MacOS/Installer)
2021-02-06 11:23:49+08 Daniels-iMac Installer[22586]: Install failed: The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance.
2021-02-06 11:23:49+08 Daniels-iMac Installer[22586]: IFDInstallController 2F0BF60 state = 8
2021-02-06 11:23:49+08 Daniels-iMac Installer[22586]: Displaying 'Install Failed' UI.
2021-02-06 11:23:49+08 Daniels-iMac Installer[22586]: 'Install Failed' UI displayed message:'The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance.'.
2021-02-06 11:23:56+08 Daniels-iMac Installer[22586]: Package Removal: User opted to keep package.
```

SergeySlice, 2021-02-06:
It looks like new Apple's security policy. Try to enable the package.

DanielYWoo, 2021-02-10:
@SergeySlice I even cannot un-zip those packages, it's not a security policy issue.

SergeySlice, 2021-02-14:
Not unzip but enable.

closed 2021-02-21 by SergeySlice