CloverHackyColor/CloverBootloader #358: Unable to load clover boot screen through refindplus chainload

jzrodriguez98, 2021-02-07:
I used to load clover boot screen through a refindplus chainload but it no longer work since clover v5128.[image]
The logs from refindplus debug versión will be attached shortly.

SergeySlice, 2021-02-07:
It is refind problem.

closed 2021-02-07 by SergeySlice

jzrodriguez98, 2021-02-07:
I opened a ticket with refindplus repo and they were willing to research it but they will need to know what was changed exactly in Clover which is causing refind not to load CLOVERx64 efi properly. Maybe you may provide some tips to them (git user is dakanji).

dakanji, 2021-02-14:
@SergeySlice 

Any hints on what might need to change on the other side?

jzrodriguez98, 2021-02-14:
@dakanji . I did a test copying refindplus version AN to my EFI and also copied Clover to the same EFI where OpenCore and refindplus files reside. REFIndPlus loaded Clover booting screen without any issues.

jzrodriguez98, 2021-02-14:
I will do another test this time copying Clover to a third EFI partition different from the one previously used.

dakanji, 2021-02-14:
if you identify that the issue is specific to a RefindPlus version since you say 0.12.0.AN works, then raise an issue back on the project and provide the pertinent information.

dakanji, 2021-04-06:
Not tested as I don't run Clover, but the issue with booting 3rd party boot managers such as Clover, should be resolved in [RefindPlus 0.13.2.AA](https://github.com/dakanji/RefindPlus/releases/tag/v0.13.2.AA).

jzrodriguez98, 2021-05-25:
@dakanji . I checked and this release did not work since it showed a white screen without boot entries to choose.

dakanji, 2021-05-25:
Please use the actual current release and raise an issue on the project page.
The issue template has the relevant information needed.
Thanks