CloverHackyColor/CloverBootloader #359: Service did not exit... then rebooting after pass first installation stage

oldman20, 2021-02-13:
[image]
[image]
Hello, clover 5219 is boot normal big sur (which is already installed with open core) but when i installation with clover5219, first stage is fine, then machine reboot, i choose install from preboot, then loading here and reboot itself. Any idea What's missing? Thank you

SergeySlice, 2021-02-14:
Something missing. Clover 5129 does loading BigSur Install.

oldman20, 2021-02-18:
> Something missing. Clover 5129 does loading BigSur Install.

yes, I think so. It's just boot from preinstall macOS Big Sur, that's all this time!

SergeySlice, 2021-02-21:
There must be more information to say something.
I just can tell that there is no proof of the Clover gilty

SergeySlice, 2021-03-16:
Abandoned

closed 2021-03-16 by SergeySlice