CloverHackyColor/CloverBootloader #360: getting a "6 _" on my boot screen when i  try to boot off of a Clover EFI bootloader USB Stick

SkyEngine-OSP, 2021-02-15:
there's really not much information but that number showing off of my screen. i tried switching boot, boot6, and boot7. none worked.

Computer Board is an **[HPe ProLiant ML310e Gen8 V2](https://support.hpe.com/hpesc/public/docDisplay?docId=emr_na-c03761153)** with an **[Eaget S900L M.2 NVMe SSD](https://www.gearbest.com/hdd---ssd/pp_009284683568.html)**

SergeySlice, 2021-02-15:
This computer should support UEFI boot. This way you will need no boot6 or other.

SkyEngine-OSP, 2021-02-15:
hmmm weird. the reason why i was trying to use Clover was primarily because my pc doesn't support native UEFI (just Legacy BIOS) and i'm trying to boot off of an NVMe drive.

tinywrkb, 2021-02-19:
I also got hit by this. This seems to be a USB-specific issue. In QEMU (syslinux) I can't boot from USB flash drive but if I convert it to a virtual SATA device then it's working.  
It's not actually an issue for me, as I can flash EFI FW for my test machine (an ancient Chromebook).  

@SkyEngine-OSP I'm not sure what you're doing but if you really need Clover then you can boot a live distro and install Clover to the SSD and try to see if it works when you're not booting from USB.  
I'm not sure what you're doing after that, what OS you're installing. Sorry I can't help further, I only care about booting Linux with my custom scripted bootloader.

SkyEngine-OSP, 2021-02-19:
@tinywrkb thank you for that. 

the reason behind this is because my Server PC, an HPe ProLiant ML310e Gen8 V2 doesn't support UEFI at all as it only has Legacy BIOS in it. my ultimate goal is to make Clover work, install Windows Server on an M2. NVMe SSD that's attached to an adapter plugged in the PCIe 3.0 x8 slot on the Motherboard and boot from it.

I'd like to make use of NVMe's big improvement in read and write capabilities versus SATA III.

i've never heard of virtual SATA nor do i know how to make them. if you can spare me a few of your time, would you be so kind as to provide a walkthrough on how? i would greatly appreciate it.

tinywrkb, 2021-02-19:
@SkyEngine-OSP the virtual SATA device was in QEMU, this is a virtualization software just like Virtualbox of VMware Workstation.  
I tested in QEMU as I suspect this might be specifically affecting USB drives but is not the best test case.  
Still, the fact I can boot Clover in QEMU when installed on a SATA drive suggests it worth trying the same on actual hardware, you won't lose much trying this except wasted time and effort.  
Though I must admit that I'm not sure why are you wasting time going with Clover? A better choice would be to install a hypervisor and run the OS in a virtual machine.  
Proxmox should be easy enough to set up by users that don't have much experience with Linux. I believe it has a web UI that allows you to stay clear from the command line and fully admin the server using a web browser.

To my other test system, an old Chromebook, I flashed UEFI FW today so no problem there anymore.

The reason I'm using Clover is to be able to run the same EFI binary that I have for regular machines/actual hardware also with VPS hosting services and maybe also as a BIOS compatibility mode for regular machines.  
For the latter use case, this is clearly a fail attempt but I care much less about it.

SergeySlice, 2021-02-21:
> hmmm weird. the reason why i was trying to use Clover was primarily because my pc doesn't support native UEFI (just Legacy BIOS) and i'm trying to boot off of an NVMe drive.
> 
> i believed Clover supports booting of EFI Drives from BIOS via UEFI Emulation, or am i wrong?

Yes, it does.
Legacy boot required NvmExpressDxe.efi. Is it here?
Are you want to show me debug.log?

SergeySlice, 2021-03-16:
abandoned

closed 2021-03-16 by SergeySlice