CloverHackyColor/CloverBootloader #361: Clover 5130 issue

himeshsam93, 2021-02-16:
I have update to latest version from clover 5120 to 5130 but  when I reboot my hackintosh it says malformed plist A fatal error happened system halted 
How to solve this problem?

stevezhengshiqi, 2021-02-16:
You certainly need to adjust your config when updating from old version Clover.
[config-sample.plist](https://github.com/CloverHackyColor/CloverBootloader/blob/master/CloverPackage/CloverV2/EFI/CLOVER/config-sample.plist) provides an example, and `Quirks` section is something new starts from Clover r5123.

himeshsam93, 2021-02-16:
Ok but I adjust my config but no luck still I can't boot into my system. can you define the quirks and drivers and all the fixes that allowed to install Big Sur , can you  upload a EFI folder that can update Big Sur . 
here the message when I boot into my machine can you tell how to fix this?
[image]

here's the my config.plist can you readjust the plist and give to me or give me a full compatible efi folder

[config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5986391/config.plist.zip)

stevezhengshiqi, 2021-02-16:
According to the error log, I will say the issue is on `Anime` dict in your theme.plist. I am not sure what exactly happens but you might need to change a theme. An upload of your theme.plist is welcome.

himeshsam93, 2021-02-16:
OK , can you please fix this to me!!!
[themes.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5986429/themes.zip)

himeshsam93, 2021-02-16:
is my config.plist is correct? if it is incorrect can you please recorrect and give me? its a huge help!

stevezhengshiqi, 2021-02-16:
Could you change a theme and test again? Anime part needs @jief666 to have a look.

himeshsam93, 2021-02-16:
hello, @stevezhengshiqi here's my EFI folder also.
[EFI.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5986479/EFI.zip)

himeshsam93, 2021-02-16:
can you give me theme.plist because I didn't have ant theme.plists

stevezhengshiqi, 2021-02-16:
theme.plist is located in your theme folder. What you can do is to change theme in config.plist - GUI - theme and test, for example, set theme value to `embedded`.

himeshsam93, 2021-02-16:
ok wait ill test it now

himeshsam93, 2021-02-16:
But no luck still it’s the same 
[image]

himeshsam93, 2021-02-16:
I have 3 themes under my theme folder in efi 
I change embeded folder :- theme.plist 
:- theme :- embedded

stevezhengshiqi, 2021-02-16:
You just need to change `config.plist - GUI - theme` to `embedded` as I said before. No other actions are needed. Or you can just delete the faulty theme folder `hackintoshzone`

himeshsam93, 2021-02-16:
Ok thank you wait I’ll check.

himeshsam93, 2021-02-16:
Yes it’s loaded the clover r5130 thank you for that 👌🏻🔥😍 but another error occurred can’t log into Mac OS 
[image]
It’s stuck on this

himeshsam93, 2021-02-16:
[image]
It’s stuck on this 
What’s the reason for this ???
Is that drivers I change drivers that are need to install Big Sur is that the problem?

himeshsam93, 2021-02-16:
is this config.plist values are correct ?
[Archive.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5986620/Archive.zip)

stevezhengshiqi, 2021-02-16:
The drivers in your kexts folder are not loaded because same kexts existed in the S/L/E or /L/E. You need to
 - update these kexts in your S/L/E (/L/E) 
 - or set `InjectKexts` to `Yes`
 - or remove these kexts in S/L/E (/L/E)

himeshsam93, 2021-02-16:
is it this?
[image]

himeshsam93, 2021-02-16:
It’s fixed thank you for helping ❤️🤩

stevezhengshiqi, 2021-02-19:
@SergeySlice Hi Slice, I can confirm that starts from r5122, the `Anime` setting in `theme.plist` would cause the error
```log
MALFORMED PLIST in 'Anime' : TagArray:dictElementAt(1) -> trying to get a dict element at 1, but element is Key
```
Relevant code should at here: https://github.com/CloverHackyColor/CloverBootloader/blob/master/rEFIt_UEFI/libeg/XTheme.cpp#L676

I don't see apparent errors and it would be nice if you can take a look.

Here's an archive of a sample theme that can cause the error by @himeshsam93 
[themes.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6006948/themes.zip)

SergeySlice, 2021-02-19:
The theme.plist is malformed
[image]
CharWidth, Path, Type must be out of Anime array.

closed 2021-02-19 by SergeySlice