CloverHackyColor/CloverBootloader #364: 5130 При установке Big Sur не отображаются внутренние диски. Только внешние носители

iAmSuperMan12, 2021-02-18:
Ноутбук HP Envy 17 j102sr, i7 4702qm, hd graphics 4600.

Использую самый актуальный загрузчик со всеми драйверами из этой же сборки.
а именно UEFI: OpenRuntime, ApfsDriverLoader.efi, VBoxHfs.efi
По итогу когда гружусь в установочник, диски не видно нигде. 
Естественно раз я загрузился в установочник, значит первично драйвера схватились и загрузчик наши диски видит. а дальше то что происходит? Куда девались?
Проблем в загрузке Mojave не выявлено кроме одной тонкости...Видеочип стал вместо 550мгц - 200мгц частота простоя.
Из за этого фризы  в системе присутствуют. 
На 5118, такого нет. Весь профиль производительности у меня прописан в SSDT.

SergeySlice, 2021-02-19:
Давай debug.log, будем смотреть кто кого не видит и с какой кстати.

iAmSuperMan12, 2021-02-19:
Прошу прощения за то, что не предоставил логи сразу.
[2021-2-19_15-2-38_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/6010929/2021-2-19_15-2-38_BOOTX64.EFI.log)

SergeySlice, 2021-02-20:
Потому что внутренние диски сидят на нестандартном контроллере
0:103  0:000  PCI (00|00:1F.02) : 8086 282A class=010400
Этот класс устройств относится к RAID, не знаю, каким образом он оказался в ноутбуке. Если есть возможность повлиять на это в БИОСе, сделай простой AHCI.
Еще попробуй легаси-загрузку, может там повезет больше.

iAmSuperMan12, 2021-02-20:
Да, это действительно так.
К сожалению я не могу препятствовать этому через Биос.
Это произошло, когда я установил ССД в mSata. Каждое включение ноутбука сопровождается уведомлением о возможности сделать рейд из подключенных жёстких дисков. И предлагает перейти в утилиту для этого. Висит сей уведомление около 5 секунд...потом происходит загрузка с носителей.

Я попытался загрузить Кловер в Легаси, но он не видит HFS разделы. Драйвера уже положил все, все, что лежали в папке BIOS/File System, но флешка с Big Sur так и не отобразилась

То есть получается что всё, Каталина это потолок для этого ноутбука?

SergeySlice, 2021-02-21:
А если и так, в чем трабл? БигСюр чем-то интересен?

iAmSuperMan12, 2021-02-21:
Пока не так чтобы очень. Эстетическое удовольствие огромное. Ещё есть подозрение, что скоро на Каталина эпл положит болт, и Logic Pro перестанет поддерживать Каталина так, как перестал поддерживать Мохаве.

SergeySlice, 2021-03-07:
Так что с этим иссью? Закрыто, как не имеющее отношения к Кловеру?

iAmSuperMan12, 2021-03-07:
Похоже что так. Странно люди как то заводят биг сюр...
Ладно, извини за то что время отнял.
Успехов во всех делах!

closed 2021-03-07 by iAmSuperMan12