CloverHackyColor/CloverBootloader #366: Clover V5.1 r5124 Intel HD 3000 Black Screen Problem

maheshkondraju, 2021-02-20:
Hi,

Intel HD 3000 Black Screen Problem (Screen completely off)  from Clover V5.1 r5124 to the latest release Clover V5.1 r5130.

Clover V5.1 r5123 + Fix TgtBridge works smoothly with out any Intel HD 3000 Black Screen Problem (Screen completely off).

As soon as i replace  Clover V5.1 r5124 BOOTX64.efi in the boot folder of EFI folder its resulting in Black Screen Problem (Screen completely off) for Intel HD 3000.

So something is missing or changed in BOOTX64.efi from Clover V5.1 r5124.

Any solution or suggestion is highly appreciated.

Regards
Mahesh Kondraju

SergeySlice, 2021-02-21:
Try to set boot-args slide=0.
It was default in 5123.

SergeySlice, 2021-03-16:
Looks like resolved as 10 days no reply.

closed 2021-03-16 by SergeySlice