CloverHackyColor/CloverBootloader #367: Remove the "X" character in label at boot screen

rex-lapis, 2021-02-23:
Hi Clover team,
First at all thank you for you hard work to make Clover better now.
I'm using Clover 5130 and it can boot into macOS Big Sur very great without any problems.
I realized that with macOS Big Sur, Clover now only shows pre-boot partition only. So I want to send a request, if you have time can you check it?
If macOS Big Sur is booted with Clover, is it possible to remove the "X" character at boot menu?, Like from **Boot Mac OS X from xxxx via Preboot** to **Boot Mac OS (or Mac OS Big Sur) from xxxx via Preboot** (because macOS Big Sur is 11 xD).
Thank you very much.
[image]

SergeySlice, 2021-02-25:
C u.

SergeySlice, 2021-02-27:
Commit 0dc0a9304

rex-lapis, 2021-02-27:
thank you for your hard work sir!

closed 2021-03-03 by rex-lapis